openscad/openscad #4781: Function realisation via classic return.

Nikita-Presnov, 2023-10-13:
Implementing the function through curly braces and return would provide much more opportunities for optimization and improving the readability of models.

jordanbrown0, 2023-10-13:
The picture is not quite as simple as it would seem.

Indeed, something like
```
function (a,b,c) {
    x = something;
    y = something;
    z = something;
    return (x + y + z);
}
```
would be more familiar than
```
function (a,b,c) =
    let(x = something, y=something, z=something)
    x + y + z;
```

But remember OpenSCAD variable scoping rules.  The only thing that could work totally "obviously" is a straight sequence of assignments.  Assignments inside `if` or `for` are not-quite-useless.
```
function f(a,b,c) {
    if (a ==b) {
        x = 1;
    } else {
        x = 2;
    }
    return (x);
}
```
won't work, because the assignments to x don't survive outside their blocks. Similarly for `for`.

To have a conditional involved in a function mostly requires using `a?b:c` syntax.

In theory `return` from the middle of an `if` or `for` could work, but that's pretty restrictive.

One other concern is the interaction with the object syntax proposed in #4478, where curly braces introduce an object.  I'm not sure that it's impossible to reconcile the two concepts, but I'm also not sure that it's possible.

pca006132, 2023-10-14:
I am not sure what additional optimization can be done with return statements. I think early return can make your code simpler (in the sense with fewer nesting depth), but it will not enable more optimizations unless you use it within a loop. But loops are not used in OpenSCAD functions because variables are immutable, you use recursion.

Nikita-Presnov, 2023-10-14:
> I am not sure what additional optimization can be done with return statements. I think early return can make your code simpler (in the sense with fewer nesting depth), but it will not enable more [...]

When formula very big.

I tried to create wing profile of Zhukovsky-Chaplygin

Nikita-Presnov, 2023-10-14:
In general, it is when you try to do something that goes beyond the usual primitives that the openscad language seems to be trying to put a spoke in your wheels.

nophead, 2023-10-14:
Just use let() for multi-line functions and ? for optional return values. You don't need to change the language, just use slightly different syntax from doing it in something like C.

Here is a fairly complex multi-line function from my library. I don't need braces or multiple return statements. 

```
unction rounded_path(path) = //! Convert a rounded_path, consisting of a start coordinate, vertex / radius pairs and then an end coordinate, to a path of points for sweep.
    let(len = len(path)) assert(len > 3 && len % 2 == 0) [
        path[0],                                        // First point has no radius
        for(i = [1 : 2 : len - 3]) let(                 // Step through the vertices with radii, i.e. not the first or last
            prev = max(i - 2, 0),                       // Index of previous point, might be the first point, which is a special case
            p0 = path[prev],                            // Point before the vertex
            p1 = path[i],                               // Vertex
            r = path[i + 1],                            // Radius of shortcut curve
            p2 = path[i + 2],                           // Point after the vertex
            v1 = assert(Len(p0) == 3, str("expected path[", prev,  "] to be a vertex coordinate, got ", p0))
                 assert(Len(p1) == 3, str("expected path[", i,     "] to be a vertex coordinate, got ", p1))
                 assert(Len(p2) == 3, str("expected path[", i + 2, "] to be a vertex coordinate, got ", p2))
                 assert(is_num(r),    str("expected path[", i + 1, "] to be a radius, got ", r))
            p0 - p1,                                    // Calculate vectors between vertices
            v2 = p2 - p1,
            a = angle_between(v1, -v2),                 // Angle turned through
            d = r * tan(a / 2),                         // Distance from vertex to tangents
            room = min(norm(v1), norm(v2)),             // Maximum distance
            arc_start = assert(d <= room,
                str("Can't fit radius ", r, " into corner at vertex path[", i, "] = ", p1, " only room for radius ", room / tan(a / 2)))
                p1 + unit(v1) * d,                      // Calc the start position
            z_axis = unit(cross(v1, v2)),               // z_axis is perpendicular to both vectors
            centre = arc_start + unit(cross(z_axis, v1)) * r, // Arc center is a radius away, and perpendicular to v1 and the z_axis.
            x_axis = arc_start - centre,                // Make the x_axis along the radius to the start point, includes radius a scale factor
            y_axis = cross(x_axis, z_axis),             // y_axis perpendicular to the other two
            sides = ceil(r2sides(r) * a / 360)          // Sides needed to make the arc
        )
        for(j = [0 : sides], t = a * j / sides)         // For each vertex in the arc
            cos(t) * x_axis + sin(t) * y_axis + centre, // Circular arc in the tiled xy plane.
        path[len - 1],                                  // Last point has no radius
    ];
```

pca006132, 2023-10-14:
can you give an actual example in the imaginary language you think of?

pca006132, 2023-10-14:
And regarding performance, I think the problem is not in the syntax but in the interpreter implementation, e.g. walking a tree structure + allocations in hot path makes the whole thing slower.

jordanbrown0, 2023-11-18:
Duplicate of #1199.

jordanbrown0, 2025-03-08:
Closed as duplicate.

closed 2025-03-09 by jordanbrown0