openscad/openscad #4784: Strange empty pop up window

Claghorn, 2023-10-16:
**Describe the bug**
I open this file from the "open recent" dialog when starting OpenSCAD:

`scale([.5,.5,1])
linear_extrude(height=3) text(str(0.88), $fn=64);
`

And a small pop up window appears over the OpenSCAD window. The contents are empty. I can dismiss it and keep
going but I have no idea why it happens.

**To Reproduce**
Described above.

**Expected behavior**
No popup.

**Code reproducing the issue**
Included above

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment and Version info (please complete the following information):**
Fedora 38 Linux from the fedora repos
OpenSCAD 2021.01
Intel PC 64-Bit

UBaer21, 2023-10-16:
Sounds like the pop up from building font cache.

CrashCash, 2023-10-19:
Just an FYI, I'm getting the same issue in Devuan. My file is just

`text("SOMETHING");`

:/> openscad-nightly -v
OpenSCAD version 2023.10.15.nightly

The window isn't actually empty, it's not refreshing the window so you can't see the contents.

jordanbrown0, 2026-04-03:
On Windows, whether or not I open a file from the command line several windows pop up and immediately disappear.  It isn't clear whether any of them are the font-cache-update window.  I used to see that all the time, but I'm not sure I ever see it now.