openscad/openscad #4785: Next official release

kjy5, 2023-10-19:
The latest release is still 2021.x and the nightlies are all the way to 2023.x. Is there a reason why some stable version has not been released in the past couple of years?

I'm particularly interested in utilizing the Apple universal binaries that have existed since 2022.x (#4036) but I don't want to use a nightly build.

Also, opening the discussion/forum tab would be useful as this post is really meant for that rather than an issue.

Thanks!

t-paul, 2023-10-19:
This is a small number of people working on this in their spare time. Creating a release takes time and at this point it's not clear when that can happen. I did the last one and people have made sure I don't have any motivation to do the next one alone.

The official place for disucssions is the mailing list as indicated on the website https://openscad.org/community.html - not everyone has / wants to have a github account so there's currently no plan on using github discussion.

closed 2023-10-19 by t-paul

kjy5, 2023-10-19:
Sounds good. Thanks for the quick reply!

t-paul, 2023-10-19:
FWIW, I would like to see a release in the not too distant future too.

edoardoo, 2024-01-15:
> I did the last one and people have made sure I don't have any motivation to do the next one alone.
>
> The official place for disucssions is the mailing list as indicated on the website [...]

Thank you for taking the time to do it, I really appreciated it!

kintel, 2024-01-15:
We'll try to track issues blocking the next release here: https://github.com/orgs/openscad/projects/2
It's not complete, but once the blocking list is almost empty, more concrete release discussions are likely going to happen :)