openscad/openscad #4788: Could the RELEASE_NOTE.md have a little update?

Sarah-C, 2023-10-23:
https://github.com/openscad/openscad/blob/master/RELEASE_NOTES.md

I know everyone's busy making things happen - so I know this won't be much of a priority....
It would be cool if someone can stick a few lines in the RELEASE_NOTES file for the development releases posted on https://openscad.org/downloads.html#snapshots 

We can see the dates change, and things are getting added and fixed and improved, but there's nowhere that it's advertised.

If there's a cool new (little) feature, or some faster rendering - that would be a fantastic reason to download a nightly build! It would even help with debugging due to having a wider test base.  =)

t-paul, 2023-10-23:
Sorry, that's not feasible unless someone joins the team taking care of that specific topic (which would be nice, but as it's not the most exciting work). The file will be updated in case there's a new release.

However, you can check progress using the following reports (based on categorized changes merged after the last release):

* [Enhancements](https://github.com/openscad/openscad/pulls?q=is%3Apr+is%3Amerged+merged%3A%3E2021-01-31+label%3A%22PR%3A+Enhancement%22)
* [Bugfixes](https://github.com/openscad/openscad/pulls?q=is%3Apr+is%3Amerged+merged%3A%3E2021-01-31+label%3A%22PR%3A+Bugfix%22)
* [Maintenance](https://github.com/openscad/openscad/pulls?q=is%3Apr+is%3Amerged+merged%3A%3E2021-01-31+label%3A%22PR%3A+Maintenance%22)

They are probably a bit out of date right now too, but I'm updating those from time to time.

closed 2023-10-23 by t-paul

t-paul, 2023-10-23:
Labels should be up-to-date now.