openscad/openscad #4790: Nested modules throw syntax error if using let.

stockholmux, 2023-10-24:
**Describe the bug**

Not sure if this is intentional but using a `let` with a nested module throws a syntax error:
```
module foo() let(
    someLet = 1
) {
    module bar() { }

    echo("foo");
}
```

Where as removing the `let` statement, the same code works fine.
```
module foo() {
    module bar() { }

    echo("foo");
}
```

Likewise, removing the nested module works fine:
```
module foo() let(
    someLet = 1
) {
    echo("foo");
}

```

**To Reproduce**

See above.

**Expected behavior**

I would expect that a nested module would work regardless of the use of `let` (or maybe throw a better error message? It seems syntactically valid)

**Code reproducing the issue**

See above

**Screenshots**
n/a

**Environment and Version info (please complete the following information):**
 - OS: Mac
 - System: arm64
 - OpenSCAD Version: 2021.01, dmg

**Library & Graphics card information**
n/a

**Additional context**
n/a

jordanbrown0, 2023-10-24:
I don't understand why, but yes, this is what the grammar says.  The top level and the top level of a module are "statement", which includes modules, functions, assignments, and module invocations, while child blocks (anything that's a child of a module invocation, notably including "if" and "let") are "child_statement" and can only have assignments and module invocations.

It sure seems to me like there's opportunities there for simplification and increased flexibility.

stockholmux, 2023-10-24:
I kind of understand why `if` would disallow nested modules: you could get into a situation where you conditionally create a module or create an overloaded module that would evaluate differently on the conditions of `if`

`module` -> `let` -> nested `module` ... puzzles me.

jordanbrown0, 2023-10-24:
No, I don't think there's really a problem there, because any module or function created would be scoped to that particular block anyway, just as assignments are scoped to that block.

You can scope functions as tight as you like, because you can create them using the "function literal" syntax, which makes them just be an assignment:
```
x = function (arg) some-expression;
```
and assignments are always allowed inside braces.

jordanbrown0, 2023-10-24:
With respect to the original comment, note that this is almost equivalent and works:
```
module foo() {
    someLet = 1;
    module bar() { }

    echo("foo");
}
```

stockholmux, 2023-10-24:
@jordanbrown0  I ended just forgoing `let` and putting it inside the module as you indicated, but I lost a little hair getting to that point.

The only reason I used `let` was that I was moving some calculations in a superfluous function into a module and copy and pasted the `let` block from the function to the module.

Anyhow, this feels like a bug since it's unexpected behaviour but probably doesn't block a lot of functionality: Low priority.

jordanbrown0, 2024-11-23:
Duplicate of #4460 .

closed 2024-11-23 by jordanbrown0