openscad/openscad #4791: The editor should allow to load and edit any text file not only scad files.

yahbluez, 2023-10-25:
While openscad did not allow scripting actions,
it is useful to run scripts in other langauages
and there for sometimes minor changes to the scripts or data files is necessary.

Would be nice to stay in the same editor.

solution:
remove the fixed file ending and allow to load any kind of text file.

jordanbrown0, 2023-10-26:
The standard Windows user interface for this is to have the file-type pulldown (in the bottom right of an Open dialog, above [Open] and [Cancel]) include "All Files (*.*)", as in this snippet from Notepad:

[image]

yahbluez, 2023-10-26:
[image]

In Linux too, but the selector is fixed to openscad files, i can't click it and choose/edit others.

OpenSCAD  version 2023.10.12
Manjaro Linux

jordanbrown0, 2023-10-26:
Yes, sorry, I was not saying that OpenSCAD did what you want on Windows.  It is the same on Windows; it only offers *.scad and *.csg.  I was trying to point out how it *should* work.

jordanbrown0, 2023-11-18:
One complexity:  OpenSCAD has some [existing special handling](https://github.com/openscad/openscad/blob/f602162d26be3ed2a4aa6e8e64ae6c5c66fb73bc/src/gui/MainWindow.cc#L295) of a few file types (stl, obj, 3mf, off, dxf, svg, amf, dat, png, json).  You can't open them using the dialog, but if you drag-and-drop one of them on OpenSCAD, or open them via the command line, the effect is to insert an `import()` or `surface()` referring to the file.

yahbluez, 2023-11-18:
Yah, that work around is be used, but it would be much more useful to remove the limitation from the file open dialog. This limitations are hand made and not the default for file open dialogs.

jordanbrown0, 2023-11-18:
Maybe the special handling could be disabled when the file is loaded via the Open dialog.

Or maybe you can't edit those few file types.  (But not being able to edit json would be a nuisance.)

Or maybe we could just get rid of that special handling.

But unfortunately [XKCD 1172](https://xkcd.com/1172/) applies.

keithel, 2024-06-13:
It is super simple to add the "All files (*.*)" filter to the open file dialog, however, as you've mentioned, some file types are treated special when loaded - even through the openFile utility function (which shows the file dialog).
For the moment I'm going to treat that as a feature to allow you to "open" stl files to add the import line for it, although maybe that should instead be initiated by some other menu item or toolbar button (File->Import perhaps).

I could implement that change right now - I know where in the code this is defined, and what it will do.
What I don't know at the moment is how the special handling is dealt with and how to circumvent it if desired, and instead just read in the file and display it in the editor.

I'm about to introduce a PR that adds an open file dialog filter for STLs, so you can show STLs in the file dialog so the user can easily select them to add import lines to their projects.

yahbluez, 2024-06-13:
Thank you.
The idea to import the path of a file as a button in the file dialog is a good idea. It belongs there and i see not the need for an additional menu entry.

Being able to open any text file also allows to open often used README files or LOG files which some may use beside of scripts to automate openscad operations.

Would like to see that your PR would accepted.

nophead, 2024-06-13:
I think you can just drag an STL file to the editor pane to get an import statement for it. Seems more logical as you can chose where it is dropped.

keithel, 2024-06-13:
@nophead that isn't how drag-import works today.
If you drag-import an STL, and land your mouse pointer at any place in the OpenSCAD window (within or outside the editor portion), it will place the import statement wherever your text cursor is located.

jordanbrown0, 2025-10-24:
The expected behavior for Windows applications is that when you ask them to open a file, they offer you files of the type that the program supports.  Allowing you to attempt to open, say, a `.png` file will not make anybody happier.  And for a long time, `.scad` files were the only files that were particularly interesting to open - and mostly, they still are.  Yes, it could have offered the `*.*` option as the second option, but it was never important enough for anybody to bother.  Somewhere in there somebody added the "automatically insert import operations" feature, and now adding the `*.*` operation would interact strangely with that feature, and in particular would still mean that you couldn't open some file types.  Could it be fixed?  Sure.  Is it important enough to anybody to spend the effort figuring out how to reconcile the behaviors?  Apparently not.