openscad/openscad #4794: Compiling issues: Manifold and CUDA

Dapid, 2023-10-26:
**Describe the bug**
Building on Fedora 38 from master, I get the following error:

```
[  9%] Building CXX object submodules/manifold/src/collider/CMakeFiles/collider.dir/src/collider.cpp.o
In file included from /home/david/gits/openscad/submodules/manifold/src/third_party/thrust/dependencies/libcudacxx/include/nv/detail/__target_macros:13,
                 from /home/david/gits/openscad/submodules/manifold/src/third_party/thrust/dependencies/libcudacxx/include/nv/target:205,
                 from /home/david/gits/openscad/submodules/manifold/src/third_party/thrust/dependencies/libcudacxx/include/cuda/std/detail/libcxx/include/__config:33,
                 from /home/david/gits/openscad/submodules/manifold/src/third_party/thrust/dependencies/libcudacxx/include/cuda/std/detail/__config:103,
                 from /home/david/gits/openscad/submodules/manifold/src/third_party/thrust/dependencies/libcudacxx/include/cuda/std/version:13,
                 from /home/david/gits/openscad/submodules/manifold/src/third_party/thrust/dependencies/libcudacxx/include/cuda/std/cstddef:20,
                 from /home/david/gits/openscad/submodules/manifold/src/third_party/thrust/dependencies/libcudacxx/include/cuda/std/type_traits:17,
                 from /home/david/gits/openscad/submodules/manifold/src/third_party/thrust/thrust/detail/type_traits.h:27,
                 from /home/david/gits/openscad/submodules/manifold/src/third_party/thrust/thrust/detail/tuple.inl:21,
                 from /home/david/gits/openscad/submodules/manifold/src/third_party/thrust/thrust/tuple.h:34,
                 from /home/david/gits/openscad/submodules/manifold/src/third_party/thrust/thrust/detail/pair.inl:23,
                 from /home/david/gits/openscad/submodules/manifold/src/third_party/thrust/thrust/pair.h:281,
                 from /home/david/gits/openscad/submodules/manifold/src/third_party/thrust/thrust/binary_search.h:26,
                 from /home/david/gits/openscad/submodules/manifold/src/utilities/include/par.h:16,
                 from /home/david/gits/openscad/submodules/manifold/src/utilities/include/sparse.h:19,
                 from /home/david/gits/openscad/submodules/manifold/src/collider/include/collider.h:17,
                 from /home/david/gits/openscad/submodules/manifold/src/collider/src/collider.cpp:15:
```
I have double-checked `MANIFOLD_USE_CUDA` is disabled in cmake, and no CUDA compiler is set.
All the dependencies seem installed:
```
(py39) [david@Hopper openscad]$ bash scripts/check-dependencies.sh 
depname           minimum           found             OKness            
qt                5.4               5.15.10           OK                
qscintilla2       2.9               2.13.4            OK                
cgal              5.4               5.5.3             OK                
gmp               5.0               6.2.1             OK                
mpfr              3.0               4.1.1             OK                
boost             1.56              1.78              OK                
opencsg           1.4.2             1.5.              OK                
glew              1.5.4             1.7.0             OK                
eigen             3.0               3.4.0             OK                
glib2             2.0               2.76.5            OK                
fontconfig        2.10              2.14.             OK                
freetype2         2.4               2.13.0            OK                
harfbuzz          0.9.19            7.1.0             OK                
libzip            0.10.1            1.9.2             OK                
bison             2.4               3.8.2             OK                
flex              2.5.35            2.6.4             OK                
make              3                 4.4.1             OK                
double-conversion 2.0.1             2.0.1             OK  
```

The changes I made on cmake:

`-DCMAKE_INSTALL_PREFIX=/home/david/.local  -DEXPERIMENTAL=true`

I am not sure what other information you need, but here is my [`CMakeCache.txt`](https://gist.github.com/Dapid/9ddae90126d50fe151e6665a9a0fcbf7).

t-paul, 2023-10-26:
What's the actual error message? The log is truncated.

OpenSCAD master branch may currently still point to a Manifold version which fails to compile with gcc-13. This is fixed upstream, update in OpenSCAD is pending in #4728.

Dapid, 2023-10-26:
Sorry, I have fixed it. The compiler must be the issue, thanks. I'll follow the other issue.

closed 2023-10-26 by Dapid