openscad/openscad #4798: Preserve original colors of minuend objects that are cut with `difference()`

amatulic, 2023-10-27:
Most of the time, I appreciate that the `difference()` operation leaves behind surfaces colored green, so it is easy to tell that those surfaces arose through that operation. It's also useful that the color of this surface can be changed by assigning a color to the subtrahend objects.

However....

### Problem
There are times when I really wish there was a way to preserve the colors of the original objects. Effectively, I want an option for the subtrahend of the `difference()` operation to have no effect on the color of the minuend.

Two examples:
* I have a complex design in which the minuend consists of multiple objects of different colors, and I want to see a cross-section view by subtracting a giant cube (the subtrahend) so that it cuts through a plane. The result is a mess of green where the cube cut through everything, made worse by the fact that some differently colored minuend objects are touching and cannot be distinguished.
* I design a pencil with a black core, brown wooden body, white primer coat, and yellow paint coat as concentric objects. If I try to "sharpen" this pencil by subtracting hollow cone from the end, I get a green pointy end, not an end that looks like a sharpened pencil.

### Solution I would like
I would like an additional optional parameter in `difference()` to suppress the coloring of the objects being subtracted from. The parameter could be `keep_original_color=false` by default, which results in the current behavior. Setting this parameter to `true` would preserve the original colors in the minuend.

### Alternatives
The only alternatives I know of would be to subtract objects individually (I tried this with the pencil). This runs into problems when the subtractions share the same surfaces (as in the two examples above), resulting in some visual scrambling during preview.

jordanbrown0, 2023-11-18:
Perhaps somebody more familiar with it than I will chime in, but absent substantial changes in how preview works I don't think this can happen.  The previewer doesn't "know" the color of the object being subtracted from.  It doesn't operate on volumes; it operates by stacking faces, with black magic to simulate subtraction.

The best chance for something like this happening is that full F6 render learns to handle colors.

amatulic, 2023-11-18:
@jordanbrown0 - well, the previewer (or something) "knows" the color of the subtrahend already (the object being subtracted), because that color is assigned to the newly cut surface. Or a default color is assigned if no color was assigned to subtrahend. If it knows the color of the subtrahend, why not also know the minuend's color, and keep the minuend's color? That's what I'm asking here.

nophead, 2023-11-18:
OpenCSG works by drawing the minuend first and then drawing the subtrahend over the top of it with appropriate graphics modes. I don't think it can preserve the minuend colour. They are all just pixels in the frame buffer.

jordanbrown0, 2023-11-18:
See my comment about "black magic" :-)

I have periodically thought that I understood how preview subtraction worked.  I don't understand it now, and I suspect that I never did.  But it's more like throwing all of the triangles at the lower layers and figuring it out in the frame buffer than it is like actually doing the subtraction.

amatulic, 2023-11-18:
@nophead - in that case, should this ticket be closed as "won't do"?

kintel, 2024-07-20:
Closing as duplicate of #492

closed 2024-07-20 by kintel