openscad/openscad #4799: Coredump with File|Quit but not Main Window Close

neildarlow, 2023-10-29:
**Describe the bug**
When quitting OpenSCAD using File|Quit are coredump results. However, using the Main Window Close icon (top-right) doesn't result in a coredump. Two distinctly different behaviours.

**To Reproduce**
Steps to reproduce the behavior:
1. Start OpenSCAD and select New
2. Click on 'Main Window Close icon'
3. OpenSCAD closes and note no coredump in /var/lib/systemd/coredump
4. Start OpenSCAD and select New
5. Click in 'File|Quit Menu item'
6. OpenSCAD closes and note a coredump in /var/lib/systemd/coredump

**Expected behavior**
File|Quit should behave like Main Window Close and not generate coredump.

**Code reproducing the issue**
If applicable, add trimmed down, self-contained code demonstrating the issue.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment and Version info (please complete the following information):**
 - OS: fedora Workstation 38 Cinnamon Edition
 - System: Intel Core i5 desktop with NVidia graphics running Nouveau driver
 - OpenSCAD Version: 2021.01 flatpak from flathub.org
Also:
 - OS: openSUSE Aeon GNOME 45
 - System: Intel Core i5 laptop with Intel iRIS Xe graphics
 - OpenSCAD Version: 2021.01 flatpak from flathub.org
 
**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

**Additional context**
Add any other context about the problem here.

sebeaumont, 2023-11-04:
Interestingly exhibits exactly the same behaviour on Mac OS Ventura on a M1 (arm64) build from MacPorts.

thehans, 2023-11-06:
This is fixed in the current source tree (since #3846).  You can run a recent build downloaded from the "developement snapshots" section here:
https://openscad.org/downloads.html#snapshots

closed 2023-11-06 by thehans