openscad/openscad #4801: Use a Fixed Width Font in Console

njsokalski, 2023-11-02:
All code & script editors should use a fixed width font in both the editor & all text output windows (such as error, debug & console windows). I normally use Visual Studio Code for editing my OpenSCAD code, but I obviously still need to have OpenSCAD open to see the preview & console. When working with arrays (which I frequently do), seeing them correctly aligned in the Console window when debugging using assert and/or echo is very important to me, but they obviously are not due to the font used by OpenSCAD. I would like to see OpenSCAD use and/or have the option to use a fixed width for the Console. I have not found a workaround, so hopefully this will be available in a future version. Thanks.

thehans, 2023-11-02:
This is already available in recent development snapshots since PR #3623

closed 2023-11-02 by thehans

njsokalski, 2023-11-02:
I tried adding consoleFontFamily & consoleFontSize to

HKEY_CURRENT_USER\Software\OpenSCAD\OpenSCAD\advanced

But it does not seem to do anything. I have restarted OpenSCAD, but no difference. I am using version 2021.01 on Windows 10 Pro.

thehans, 2023-11-02:
There has not been an official release since the change was made, so configurable console font is not available in the 2021.01 Release.  

But you can run a "development snapshot" which you can find on the downloads page: https://openscad.org/downloads.html#snapshots
These include many new features and fixes, since these snapshots are kept current with the github source repository.

njsokalski, 2023-11-03:
Thanks, I think that solved the problem, except that even though the keys are saved in the Registry & displayed in Preferences, it does not apply that Preference in OpenSCAD until you go to Preferences and select it again. But I can deal with that for now, thanks for the info!