openscad/openscad #4802: OpenSCAD Intermittently Crashing When Making Some Simple Shapes (Mac 14.1)

mindlessroman, 2023-11-02:
**Describe the bug**
Doing some exercises to get familiar with OpenSCAD, I was writing the code for a simple cylinder that would eventually union another cylinder. Start typing the second cylinder code and the app crashes.

**To Reproduce**
Steps to reproduce the behavior:
1. Start typing the code below.
2. Observe crash

**Expected behavior**
I guess to not crash?

**Code reproducing the issue**
```scad
cylinder(9, 10, 0);
cylinder(9, 0, 10); // Will usually crash with an open parenthesis it seems?
```

**Screenshots**
[image]

**Environment and Version info (please complete the following information):**
 - OS: Mac Sonoma 14.1
 - System: Intel Core i9 ?
 - OpenSCAD Version: 2021.01 
   - First via `brew install openscad`, then uninstalled.
   - Then from the `.dmg` from downloads page on the site.

**Library & Graphics card information**
<details>
<summary>My "Lib and Build Info"</summary>
```txt
OpenSCAD Version: 2021.01
System information: Mac OS X Version 14.1 (Build 23B74) MacBookPro16,1 8 CPUs 16.00 GB RAM 
User Agent: OpenSCAD/2021.01 (Mac OS X Version 14.1 (Build 23B74) MacBookPro16,1)
Compiler: Clang "12.0.0 (clang-1200.0.32.28)" 64bit
MinGW build: No
Debug build: No
Boost version: 1_74
Eigen version: 3.3.7
CGAL version, kernels: 4.14.3, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.9.9
QScintilla version: 2.11.6
InputDrivers: 
GLib version: 2.56.3
lodepng version: 20180910
libzip version: 1.5.1
fontconfig version: 2.13.1
freetype version: 2.9.1
harfbuzz version: 2.3.1
cairo version: 1.16.0
lib3mf version: 1.8.1
Application Path: /Applications/OpenSCAD.app/Contents/MacOS
Documents Path: /Users/USERNAME/Documents
User Documents Path: /Users/USERNAME/Documents
Resource Path: /Applications/OpenSCAD.app/Contents/Resources
User Library Path: /Users/USERNAME/Documents/OpenSCAD/libraries
User Config Path: /Users/USERNAME/Library/Application Support/OpenSCAD
Backup Path: /Users/USERNAME/Documents/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
/Users/USERNAME/Documents/OpenSCAD/libraries
/Applications/OpenSCAD.app/Contents/Resources/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
/usr/share/fonts
/Users/USERNAME/.local/share/fonts
/Users/USERNAME/.fonts
/usr/X11/lib/X11/fonts
/System/Library/Fonts
/Library/Fonts
/Users/USERNAME/Library/Fonts
/System/Library/Fonts/Supplemental
GLEW version: 2.1.0
OpenGL Version: 2.1 ATI-5.1.35
GL Renderer: AMD Radeon Pro 5500M OpenGL Engine
GL Vendor: ATI Technologies Inc.
RGBA(8888), depth(0), stencil(0)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Qt graphics widget: QOpenGLWidget
QSurfaceFormat: RGBA(888-1), depth(-1), stencil(-1)

GL Extensions:
GL_APPLE_aux_depth_stencil
GL_APPLE_client_storage
GL_APPLE_element_array
GL_APPLE_fence
GL_APPLE_float_pixels
GL_APPLE_flush_buffer_range
GL_APPLE_flush_render
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_APPLE_pixel_buffer
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_specular_vector
GL_APPLE_texture_range
GL_APPLE_transform_hint
GL_APPLE_vertex_array_object
GL_APPLE_vertex_array_range
GL_APPLE_vertex_point_size
GL_APPLE_vertex_program_evaluators
GL_APPLE_ycbcr_422
GL_ARB_color_buffer_float
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_instanced_arrays
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_seamless_cube_map
GL_ARB_shader_objects
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_shadow_ambient
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_transpose_matrix
GL_ARB_vertex_array_bgra
GL_ARB_vertex_blend
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_window_pos
GL_ATI_blend_equation_separate
GL_ATI_blend_weighted_minmax
GL_ATI_separate_stencil
GL_ATI_texture_compression_3dc
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_bindable_uniform
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_depth_bounds_test
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture_array
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_s3tc
GL_EXT_texture_env_add
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_mirror_clamp
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_IBM_rasterpos_clip
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_light_max_exponent
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SGI_color_matrix
</details>

**Additional context**
Add any other context about the problem here.

thehans, 2023-11-02:
Can you please test on a recent development snapshot
https://openscad.org/downloads.html#snapshots

Latest build for Mac as of today is 10/30/2023 https://files.openscad.org/snapshots/OpenSCAD-2023.10.30.dmg

mindlessroman, 2023-11-03:
Oh my goodness, what a turn-around using the dev build. Thus far no crashes 🤞

mindlessroman, 2023-11-03:
Okay, slight update, did have a sudden crash when saving a different design. Again a finite number of objects in the code.  But it didn't immediately crash just while writing so that is a huge improvement.

kintel, 2023-11-03:
Thanks for the update! Unfortunately, I haven't managed to reproduce this on a M1 VM running macOS 14.1.

I'm wondering if this is related to macOS 14.1, perhaps in combination of running on Intel.

Do you happen to have access to any other Macs running macOS 14? It would be interesting to see if you're able to reproduce on Apple Silicon or if it's isolated to Intel.

mindlessroman, 2023-11-06:
I'm seeing if any of my peers see this - will report back. It happened again with the 2023.10.30 build, and it seems to happen when I'm making the open curly bracket keystroke `{`?? It's very bizarre

mindlessroman, 2023-11-20:
Sorry for the delay here! I haven't gotten anyone with a Apple Silicon chip. But I'm still seeing the error on 2023.11.05... I grabbed the crash report and [put it in a gist](https://gist.github.com/mindlessroman/c2abfec93fb776a101804e81919c891c). Hopefully this is helpful? Let me know if I can provide anything else @kintel

kintel, 2023-11-21:
Thanks for the update. Based on your description and crash report, this sounds like a UI bug, which isn't something that's automatically tested. So far, your report is the only one of this kind, so my current hypothesis is still that this is a problem triggered by the combination of macOS 14 and Intel. 

Thinking loudly in case anyone else is reading this:
* This could be a bug in Qt. Search around to see if anyone else has issues with Qt5 (5.15.7) and macOS 14 on Intel
* Qt5 isn't supported on macOS 14 (neither is Qt6, but will be in the next release). Perhaps time to bite the bullet and upgrade to Qt6.
* Perform some manual testing on an older Intel Mac to see if this is reproducible there.
* Consider building a debug build to get a more readable crash reports.

kintel, 2026-08-29:
Closing as we have long since upgraded to Qt6

closed 2026-08-29 by kintel