openscad/openscad #4804: Save Preset button is missing.

rbielaws, 2023-11-03:
I don't/can't build my own version so I pulled the pre-compiled 2023.10.27 (git ab5dd9338) a few days ago and ran it on an Intel 64bit Windows 10 machine.

I don't see the button to save presets nor any indicator that presets have changed and therefore need saving.
[image]

Both creating them and using them if they exist seem to work but the only way I could find to create/update presets via the Gui is to exit OpenSCAD.  This results in a message "Some tabs have unsaved changes..." to which I can select [Save All] to get my presets created/saved.  

I'm not sure if it's a separate issue but the message itself seems misleading because code appears on tabs implying the message is referring to a code tab.  The presets don't appear on a tab.

jordanbrown0, 2023-11-03:
File/Save seems to save both the .scad file and the .json file that contains the presets.

There used to be a separate [save preset] button but it was removed in PR#3755 "Customizer rework".

The presets do not appear on a tab, exactly, but they *are* tied to a particular tab.

t-paul, 2023-11-04:
Plus you get the `*` modified marker in the window title when changing the presets, indicating the state.

So it's not a bug. If it's the best option is debatable, I think it's ok treating the parameters as belonging to the same unit of work even though it's saved in 2 files.

closed 2023-11-04 by t-paul

rbielaws, 2023-11-04:
Ok, so it will be covered in docs.  A feature, not a bug :-).

But this answer seems to open a can of worms.  
I use Emacs so I have no editor window open.  This leaves me with no way to tell if settings are unsaved on a 'tab' that isn't in focus.  Moreover without an editor window, I see no way to switch render/customizer focus from one source to another because the 'tabs' disappear if the window is closed.  If the Next/Previous Window options on the Window menu 'should' do this, currently they don't and they don't seem to have associated hotkeys making it very inconvenient if they 'should' work.

As a workaround to (2) I've docked the internal editor in a very minimal size so that the tabs are available to switch between files and see the modified (*) indicator but I hope this isn't considered the 'solution' and doesn't help with the lack of [revert].

Also, THANKS: 
I noticed the problem where if I hit save in the OpenSCAD GUI, despite editing in Emacs no longer makes me revert buffer or force overwrite in Emacs.  Not updating the timestamp if nothing needs saving has long been desired.
Also, if I've saved in Emacs, I can't even edit from the GUI editor anymore.  I assume that's deliberate.  I think I like it.  Much harder to get mixed up and have unsaved edits in both windows.  Kudos!

jordanbrown0, 2023-11-04:
I wonder if the tabs really belong on the 3D view pane, not the editor pane.

rbielaws, 2023-11-06:
Hard to disagree with that.
I was thinking the same.

Rick

giuarb, 2024-01-21:
save preset was more efficient and intuitive

giuarb, 2024-01-21:
any change modifies the current preset even without manually save it, I am duplicating each preset to avoid that so i can experiment with parameters without loosing the original preset