openscad/openscad #4807: No

rbielaws, 2023-11-04:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Code reproducing the issue**
If applicable, add trimmed down, self-contained code demonstrating the issue.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment and Version info (please complete the following information):**
 - OS: [e.g. Windows 10, macOS 12 Monterey, Ubuntu 22.04]
 - System: [e.g. Intel PC 64-bit, Raspberry PI 4]
 - OpenSCAD Version [e.g. 2021.01, 2023.04.16] and where it was installed from

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

**Additional context**
Add any other context about the problem here.

closed 2023-11-04 by thehans

rcolyer, 2023-11-04:
ctrl-z and ctrl-shift-z already do this.

rbielaws, 2023-11-04:
I guess I'll open an actual bug report then.  I thought I tried those keys before asking.  It didn't work but I didn't know for sure that it should.  Now that you tell me it should, I can be more thorough in reporting it.

t-paul, 2023-11-04:
I'm pretty sure the field wise undo is a feature of the GUI framework. If that would not work for you, there's probably not much we can do on OpenSCAD side.

I can see a possible feature request allowing an "revert to saved state" per parameter set. I don't know how difficult that is to fit into the current customizer but this seems like something that could be useful.

rbielaws, 2023-11-04:
I've no idea what you mean by "field wise undo .. not work for you".  since there seems to be no way to undo anything, field-wise, set wise, or even the entire json.  Nothing works.  Even this would be fine if you were not, essentially forced, to save.  There is no way to continue working without first saving.  Your only choice seems to be exit, answer [revert] to warning (which under circumstances of forced exit like auto-reboot is impossible) and restart the program, re-open your files...

I even tried creating a new set before saving and the result was both sets had the same updated content.  Once you've made ANY change there is NO practical way to get the old values back.  I believe that [revert] simply re-loaded all values from json.  I don't know if there were complaints but that's far better than TOO BAD your stuck with it.

jordanbrown0, 2023-11-04:
For me, Ctrl+Z works in some cases.
* Only on the field the focus is in.  (Change A, focus B, Ctrl-Z, does nothing.  Then change B, focus A, Ctrl-Z reverts A.)
* In 2021.01, only on simple number fields, not on text fields.
* Sliders it works for changes made in the text box, not for change made with the slider.
* Does not undo number changes made with spinner arrows.

So it seems like there's some work that could be done here.  (But I don't know how much is under our control, vs under the GUI infrastructure's control.)

jordanbrown0, 2023-11-04:
On Windows.

rbielaws, 2023-11-04:
That is a GIANT leap backward.  And it doesn't even mention the [-] button which, if hit accidentally, tosses everything in the set without recourse.  Basically the new 'feature' almost completely destroys the usability of sets by forcing you to save even highly destructive changes.  If that's how it is intended to work I think most people (faced with full knowledge of the impact) would want the [revert] button back.  At least it's not destructive.