openscad/openscad #4809: Export multiple volumes from a single script

kvtorp, 2023-11-04:
With the proliferation of multi-color and multi-material 3D printers, I am starting to miss a feature to export multiple volumes from a single file, for later import into the slicer.
Specifically, the feature should:
* Trigger a separate export (either as a separate object in multi-object files suchs as 3MF, or as separate files for formats such as STL)
* Accept a label argument, used for naming the object in the export (3MF) or as suggested file name (STL)
* Prevent the indicated volumes from being automatically joined with any other parts of the model they touch (see example below)

So that a model like this:
```
volume("bottom") {
  difference() {
    Base();
    Modifier();
  }
}

volume("top") {
  intersection() {
    Base();
    Modifier();
  }
}

module Base() {
  cube(10);
}
module Modifier() {
  translate([-5, -5, 5])
    cube(20);
}
```
Would import into the slicer like this:
[image]
[image]

I am suggesting volume(), as it would directly affect the volume count at render.

t-paul, 2023-11-04:
Duplicate of #1608.

closed 2023-11-04 by t-paul

kvtorp, 2023-11-04:
Argh, I didn't find it when I searched... Yes, it is a duplicate of the parts() operator