openscad/openscad #4815: Windows 2023-08.25 crash on salt pig lid model with manifold enabled

owebeeone, 2023-11-09:
**Describe the bug**
Attempting to render the model linked below crashes or wedges Openscad v Windows 2023-08.25

https://drive.google.com/file/d/1qmgVoY0NuFzueMCYkebTCH9kjReEhAtO/view?usp=sharing

**To Reproduce**
Steps to reproduce the behavior:
1. Load the model linked
2. Render

**Expected behavior**
Create a model

**Code reproducing the issue**
(https://drive.google.com/file/d/1qmgVoY0NuFzueMCYkebTCH9kjReEhAtO/view?usp=sharing)

**Environment and Version info (please complete the following information):**
 - OS: Windows 11
 - System Model	ROG Strix G513RM_G513RM
 - Installed Physical Memory (RAM)	32.0 GB
 - OpenSCAD Version [2023-08.25 ] installed exe installer

**Library & Graphics card information**
OpenSCAD Version: 2023.11.08 (git e7abd10a9)
System information: Microsoft Windows 11 (10.0.22621) x86_64 16 CPUs 31.26 GB RAM
User Agent: OpenSCAD/2023.11.08 (git e7abd10a9) (Microsoft Windows 11 (10.0.22621) x86_64)
Compiler: GCC "11.3.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_81
Eigen version: 3.4.0
CGAL version, kernels: 5.5, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Qt version: 5.15.10
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.70.2
lodepng version: 20210627
libzip version: 1.5.2
fontconfig version: 2.14.2
freetype version: 2.13.1
harfbuzz version: 7.3.0
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: fast-csg, fast-csg-safer, fast-csg-debug, manifold*, roof, input-driver-dbus, lazy-union*, vertex-object-renderers, vertex-object-renderers-indexing, vertex-object-renderers-direct, vertex-object-renderers-prealloc, textmetrics, import-function, predictible-output
Application Path: C:/Program Files/OpenscadDev
Documents Path: C:\Users\gianni\Documents
User Documents Path: C:\Users\gianni\Documents
Resource Path: C:/Program Files/OpenscadDev
User Library Path: C:/Users/gianni/Documents/OpenSCAD/libraries
User Config Path: C:\Users\gianni\AppData\Local/OpenSCAD
Backup Path: C:/Users/gianni/Documents/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
C:/Users/gianni/Documents/OpenSCAD/libraries
C:/Program Files/OpenscadDev\libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
C:/WINDOWS/fonts
C:/Users/gianni/AppData/Local/Microsoft/Windows/Fonts
C:/Users/gianni/.local/share/fonts
C:/usr/local/share/fonts
C:/usr/share/fonts
C:/usr/X11/lib/X11/fonts
C:/System/Library/Fonts
C:/Library/Fonts
C:/Users/gianni/Library/Fonts
GLEW version: 2.1.0
OpenGL Version: 4.6.0 Compatibility Profile Context 22.20.28.09.221101
[... 239 lines cut ...]
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_storage
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array
GL_KHR_blend_equation_advanced
GL_KHR_blend_equation_advanced_coherent
GL_KHR_debug
GL_KHR_no_error
GL_KHR_parallel_shader_compile
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_NV_blend_square
GL_NV_copy_image
GL_NV_primitive_restart
GL_NV_shader_atomic_int64
GL_OES_EGL_image
GL_OES_draw_texture
GL_OVR_multiview
GL_OVR_multiview2
GL_OVR_multiview_multisampled_render_to_texture
GL_S3_s3tc
WGL_EXT_swap_control

owebeeone, 2023-11-09:
This is a low res version of the model above, it also crashes - https://drive.google.com/file/d/1RRS6CzdnHLmmJMYyWSKS25_chYqjrkWP/view?usp=drive_link

gsohler, 2023-11-09:
it does not crash on my
OpenSCAD <https://www.openscad.org/> version 2023.10.22 (git )
in windows

but it appears  that there are some volumes missing ?
the volume is  very thin on the right hand side and probably no printer can
easily resolve that.

https://imgur.com/a/W3SGsWB

owebeeone, 2023-11-09:
It definitly prints. https://makerworld.com/en/models/18828#profileId-17591

pca006132, 2023-11-09:
is it related to #4814?

owebeeone, 2023-11-11:
Another data point. I decided to roll back openscad to snapshot version 2023.10.27 and it renders properly and very fast.
OpenSCAD Version: 2023.10.27 (git ab5dd9338) - I turned on manifold and lazy-union exactly as before 11.08.

kintel, 2023-11-11:
I added the low-res version as a test case, and for some reason, it doesn't crash: https://github.com/openscad/openscad/pull/4819

@owebeeone Could you try reproducing on the cmd-line?
`openscad.com issue4815.scad --enable=manifold --render -o out.png`

issue4815.scad is https://github.com/openscad/openscad/blob/1ff854259dcd524723bffec272d61395734e6ad5/tests/data/scad/3D/issues/issue4815.scad

owebeeone, 2023-11-12:
@kintel I have reproduced the error wirh 11.08 with the command line you suggest. openscad terminates with -o out,png without producing a png file. Sometimes it will segv. When I try to render stl (out,stl) it will sometimes hang.

This is the output with various command lines:
- works 
$ cmd /c 'C:\\Program Files\\OpenSCADDev\\openscad.exe' issue4815.scad -o out.png

Compiling design (CSG Products normalization)...
Normalized CSG tree has 35 elements
Geometries in cache: 18
Geometry cache size in bytes: 141656
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0:00:00.133

Added --render
$ cmd /c 'C:\\Program Files\\OpenSCADDev\\openscad.exe' issue4815.scad --render -o out.png

Geometries in cache: 64
Geometry cache size in bytes: 888944
CGAL Polyhedrons in cache: 12
CGAL cache size in bytes: 8549560
Total rendering time: 0:00:03.187
Top level object is a 3D object:
   Simple:     yes
   Vertices:     1304
   Halfedges:    5214
   Edges:        2607
   Halffacets:   2586
   Facets:       1293
   Volumes:         2

Failures:
$ cmd /c 'C:\\Program Files\\OpenSCADDev\\openscad.exe' issue4815.scad --enable=manifold  --render -o out.png

 Got a Dr watson - segv reading 0x18

- repeat
$ cmd /c 'C:\\Program Files\\OpenSCADDev\\openscad.exe' issue4815.scad --enable=manifold  --render -o out.png

Segmentation fault  (...cygwin segv...)

- repeat 2
$ cmd /c 'C:\\Program Files\\OpenSCADDev\\openscad.exe' issue4815.scad --enable=manifold  --render -o out.png

- just hangs there, wedged, doesnt exit

- try to render STL
$ cmd /c 'C:\\Program Files\\OpenSCADDev\\openscad.exe' issue4815.scad --enable=manifold  --render -o out.stl

- Also just hangs
- repeat - segv reading 0x18
- repeat - exits without output and a status of 127
$ echo $?
127
- repeat - just hangs again
These are the 4 error modes after further repeated attempts in random order. segv 0x18, cygwin segv, exit status 127 and wedge,

Just in case, this is the sha256sum of issue4815,scad since I copied it from anchorcad_HingedLid_example2_example.scad
e2038911ec498b43f87b8526d7164400cf2dc9235f0553d1523f437280bb61e6 *issue4815.scad

OpenSCAD version 2023.10.27 on the other hand works just fine:
$ cmd /c 'C:\\Program Files\\OpenSCADDev\\openscad.exe' issue4815.scad --enable=manifold  --render -o out.stl

Geometries in cache: 76
Geometry cache size in bytes: 888944
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0:00:00.998
Top level object is a 3D object:
   Facets:       2420

kintel, 2023-11-14:
@owebeeone Please give this build a spin: https://app.circleci.com/pipelines/github/openscad/openscad/4212/workflows/56748a42-6ca7-47b9-9372-66ba2a4d17ec/jobs/16991/artifacts

pca006132, 2023-11-14:
Hi, have you been able to make manifold work on simpler models? I feel like this is a build issue rather than bug with certain model.

owebeeone, 2023-11-15:
Tried it on OpenSCAD version 2023.11.15, appears to work fine now.

owebeeone, 2023-11-15:
Tried Version 2023, 11, 15, 18, 23, 34, the salt pig lid model works fine. Running against all other anchorscad models too. Will report any issues.

pca006132, 2023-11-15:
by 2023.11.15, do you mean the build in #4830?

owebeeone, 2023-11-15:
@pca006132 Yes, the one on #4830

t-paul, 2023-11-23:
Added latest oneTBB to MXE, build fixed via #4847. Builds will upload in the next couple of hours.

closed 2023-11-23 by t-paul