openscad/openscad #4817: OpenScad adds a notch where there is none.

threed-factory-store, 2023-11-10:
**Describe the bug**
When I run "openscad -o ..." on the attached scad file, it generates an stl file with a notch on the bottom right of the model.

**To Reproduce**
See above.

**Expected behavior**
No notch.

**Code reproducing the issue**
See attached files

**Screenshots**
[image]
[image]

**Environment and Version info (please complete the following information):**
 - OS: Ubuntu 23.10
 - System: Intel PC 64-bit
 - OpenSCAD Version 2023.10.12

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

**Additional context**
Add any other context about the problem here.

[OpenScadExample.zip](https://github.com/openscad/openscad/files/13319174/OpenScadExample.zip)

thehans, 2023-11-10:
Example is too long and missing code at the same time. 

```
Parsing design (AST generation)...
[WARNING: Can't open library '/home/walt/Documents/MyCameraGrip/OpenScad_Prod/fillets3d.scad'. in file 20231109_150803_CameraRequest_walt+admin@wzjz.us.scad, line 1](https://github.com/openscad/openscad/issues/1,/home/hans/.cache/.fr-SbjHgJ/20231109_150803_CameraRequest_walt+admin@wzjz.us.scad)
Compiling design (CSG Tree generation)...
[WARNING: Ignoring unknown module 'bottomChamfer' in file 20231109_150803_CameraRequest_walt+admin@wzjz.us.scad, line 6](https://github.com/openscad/openscad/issues/6,/home/hans/.cache/.fr-SbjHgJ/20231109_150803_CameraRequest_walt+admin@wzjz.us.scad)
Compiling design (CSG Products generation)...
```

Please make some effort to reduce the size of your example code, removing parts which do not affect the issue you are seeing, until you can't remove any more.  This would help to narrow down what exact operation or combination is the heart of the issue.

threed-factory-store, 2023-11-11:
Sorry, here's fillets3d.scad.   I renamed it so it would upload.

[fillets3d.json](https://github.com/openscad/openscad/files/13325352/fillets3d.json)

I'll see if I can simplify the model.

threed-factory-store, 2023-11-11:
Ah.  Looks like it's in the fillets3d library.  I removed the bottomChamfer() call on line 6 of the scad file and the notch disappeared.

pca006132, 2023-11-12:
so I guess this issue should be closed?

closed 2023-11-12 by threed-factory-store