openscad/openscad #4823: documentation lacking for offset()

purple512, 2023-11-12:
**Describe the bug**
One of the example files that you ship with OpenSCAD (GEB.scad) contains expressions like `offset(0.3)`.

I'm trying to comprehend the [documentation for `offset()`](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Transformations#offset) and it talks about `r` and `delta` modes, but every example in the documentation uses a named parameter! Nothing tells me which mode is used in `offset(0.3)`. I'm guessing `r` because it was mentioned first? But who knows.

Why is this so unnecessarily obfuscated? Why can it not say "r. double. The default type if not specified."? Or something along those lines?

Also, this documentation is confusing because it makes it look at first glance like the function takes three parameters, but there's really only two.

[image]

It would be much more clear at a glance if you included the function signature from the cheat sheet: `offset(r|delta,chamfer)`.

jordanbrown0, 2023-11-12:
Somewhat unfortunately, the documentation on wikibooks is actually more accurate.  If you want to use positional parameters, they are `r`, `delta`, and `chamfer`, in that order - and if you supply a valid number for `r`, `delta` is ignored.

If you want delta mode, the best strategy is to set delta as a named parameter `offset(delta=10) ...`, and then if you want chamfer mode set that as a named parameter too.  The implementation currently allows you to pass a non-number for `r` and it will be ignored, `offset(undef, 3, true)`, but I would not recommend it - that seems like the kind of pattern that might yield a warning in the future.

We should probably change the cheat sheet to `offset(r, delta, chamfer)` and clarify in wikibooks that if you want delta mode you should use named parameters.

amatulic, 2024-02-06:
I have fixed the documentation at https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Transformations#offset to address the issues raised in this ticket. If that is satisfactory, this issue may be closed.

I submitted a related issue #4973 regarding the tooltip that pops up for offset().

jordanbrown0, 2024-02-08:
Thanks for the cleanup.  Works for me.

closed 2024-02-08 by jordanbrown0