openscad/openscad #4824: catch and report (manifold)

UBaer21, 2023-11-12:
**Environment**
v. 2023.11.12 on Win10

**Issue**
scad crash/terminated sometimes without warning or error when using manifold.

command line with debug sometimes (not every crash) gave 
[image]

Note: 
preview and render should catch all generated exceptions at the top level

kintel, 2023-11-12:
@pca006132 Is there a good place to open this discussion in the context of Manifold? Should Manifold guarantee to not throw any exceptions (apart from `std::bad_alloc`), or should client code always wrap Manifold calls in try-catch blocks?

pca006132, 2023-11-12:
For mesh to manifold, it is expected to throw some error if the input is not a valid manifold. Otherwise, we should not throw any exceptions (unless processOverlap is false and MANIFOLD_DEBUG is ON, but I think OpenSCAD is not using that right now). Note that a seemingly weird behavior is that the exception usually happen when you eventually evaluate the CSG tree, i.e. when you run `ToMesh` or something that triggers the evaluation, due to the lazy behavior of Manifold.

I think we can open an issue in manifold repo. We should improve the documentation about this.

jordanbrown0, 2023-11-12:
A relevant question is how you want OpenSCAD to respond to bugs in itself, either bugs in OpenSCAD per se or in Manifold.

Absent any bugs, exceptions that are not explicitly documented will not happen and so are not an issue.  (For instance, "input is not manifold" might be an expected exception.)

It is only when there is a bug that an unexpected exception will occur.  At that point you have three-ish choices:  (a) you can attempt to continue with the render, (b) you can abort the render but have the program continue to operate, or (c) you can crash the program and produce a core dump for later analysis.

I'm a core dump fan, because I like forensic data and am terrified of silent corruption.  However, in this case where silent corruption isn't much of an issue, aborting the render might be reasonable.  I wouldn't attempt to continue with the render; it seems almost certain not to produce the desired results and any results that it does produce will confuse the issue.

kintel, 2023-11-12:
Taking down our IDE isn't cool though. We might be able to rearchitect stuff to always run geometry ops in separate process, but I don't see that happening anytime soon.

jordanbrown0, 2023-11-12:
I'd say that it depends on how frequent such a crash is.  Automatically saving before renders helps a lot with the impact of a crash, and a mechanism that automatically picked up the pieces on restart would help more.  If it crashed every day, that would be pretty annoying.  Every month... not sure.  Every year, no big deal.

Offsetting the inconvenience is the diagnosability; for unreproducible failures a core dump may make diagnosis easy while user reports might leave it as a mystery.

kintel, 2024-06-23:
@UBaer21 Lots of fixes has happened since you reported this. Have you experienced anything like this lately, or can we close this as fixed?

UBaer21, 2024-06-23:
I do get often a crash when open openSCAD put in a simple cube(50); and hit F6 ↦ it just closes, next time i try it works - hard to tell the reason probably windows. And it seems not when open existing designs. But i am good to close this one.

UBaer21, 2024-06-23:
Closed as not reasonable to keep open after that many changes.

closed 2024-06-23 by UBaer21