openscad/openscad #4825: Move manifold out of experimental

kintel, 2023-11-12:
This is a meta issue collecting everything needed or wanted to be able to move Manifold out of experimental and make it the default/preferred/only geometry backend in OpenSCAD.
We may want to ship Manifold before all of these issues are addressed, in which case we should split out the unsolved topics into separate issues.

* [ ] Go through all export options and update code to work without CGAL if possible
* [ ] Look through the 2D backend to make sure it's well understood how it interacts with Manifold
* [ ] Consider moving all our Clipper V1 code to Clipper V2 and use the same Clipper build as Manifold
* [ ] Look into our libtess usage and see if it's possible to use the same tessellator as Manifold
* [ ] Find a way of running a larger scale test of OpenSCAD with CGAL vs. Manifold backends to attempt to exercise more cases than the test framework currently does (e.g. download some SCAD collections and run against both backends)
* [ ] Consider making the CGAL and Manifold backends co-exist and add UI + cmd-line features to choose which one to prefer
* [ ] Make sure Manifold failures and exceptions don't crash the OpenSCAD process: #4824
* [ ] Track missing Manifold features which currently requires a CGAL fallback
* [ ] Manifold end game: Be able to build _without_ CGAL without loss of functionality

pca006132, 2023-11-12:
I think one important missing feature would be minkowski sum. We are not very interested in implementing the convex decomposition algorithm required because we think there are better algorithms for 3D offsetting, but we do accept PRs if anyone is willing to do it.

kintel, 2023-11-12:
Good point about minkowski. A very common use-case for minkowski is indeed 3d offset, but there may be other use-cases for which 3d offset wouldn't suffice. We're kind of committed to supporting minkowski forever, but we could use Manifold for the resulting large union of convex objects (unless we already do that, and the CGAL API gives us easy access to those objects). We also have a special-case already which optimizes the case for which all minkowski operands are convex.

kintel, 2023-11-12:
Updated text to mention minkowski

pca006132, 2023-11-13:
I think the current minkowski already uses manifold if possible: https://github.com/openscad/openscad/blob/master/src/geometry/manifold/manifold-applyops-minkowski.cc#L196

ochafik, 2023-11-21:
> I think the current minkowski already uses manifold if possible: https://github.com/openscad/openscad/blob/master/src/geometry/manifold/manifold-applyops-minkowski.cc#L196

Yes it does, and hopefully soon it will also use Manifold's Hull (incubating this [here](https://github.com/openscad/openscad/compare/master...ochafik:openscad:manifold-hull) BTW)

The convex decomposition is a tough one. And currently if it fails we also use CGAL's Nef minkowski as a fallback, even when Manifold backend is enabled.

TBH I wouldn't necessarily make it a strict goal to wean ourselves off of CGAL. We could just progressively restrict its use to what it's still the best at (e.g. use it to [repair](https://github.com/openscad/openscad/compare/master...ochafik:openscad:repair) meshes or for minkowski), and ensure it progressively becomes the exception rather than the rule.

@kintel To answer your open question from https://github.com/openscad/openscad/pull/4646#event-11018650929, I think it might soon be time to promote Manifold to the default (if only, to increase the amount of feedback / bug reports from users), and decide what should be the fallback behaviour: we could have a `no-manifold` experimental feature that uses fast-csg or Nef, depending which one we think is most preferable / backwards compatible (as a first step we could keep both of course). Maybe the fallback could be a non-experimental flag, so we could ship a release with the two backends if we really fear regressions. In all cases some Nef will remain necessary for minkowski for a while, so if binary size is a factor we could just save a lot by having just Manifold + Nef (+ some bits of surface mesh for repair purposes).

Also, not sure if we also need some OpenGL work to make Manifold (or fast-csg) a 1st class citizen, I remember cutting some corners / converting to PolySet or CGAL_Polyhedron a bit aggressively there.

kintel, 2023-11-21:
For the record, I agree that completely weaning off CGAL isn't an explicit goal, but being able to isolate it could be a way of improving code health. Both your suggestions are already in my list above. Feel free to edit/reword/split into separate issues once you think anything is well-defined enough to explore as separate efforts.

The rendering pieces may benefit from waiting for #4782

pca006132, 2023-11-28:
Btw there are some weird issues regarding geometry cache, #4732 and #4753.

kintel, 2023-11-28:
We have similar geometry cache issue for CGAL. I believe the issue is that we can cache geometry as Nef polyhedrons or PolySets, but since PolySets are lossy, correctness of the final result may depend on avoiding cached PolySets. ..and sometimes, doing preview before render primes the cache with non-Nef3 data, or smth.
I don't think we have a cache for Manifold objects, so we convert to/from PolySet for caching purposes, which may also be lossy. I haven't read the Manifold-related code in detail.

nophead, 2023-12-02:
Clipper can fix self intersections if you offset by a minute amount. I have never tried 0, so I don't know if that works. It depends if it gets ignored completely or is still passed through clipper.

pca006132, 2023-12-02:
The problem is that if manifoldness (each point is connected to 2 other points) is maintained and we keep the exact coordinates, the polygon will be self-intersecting. Just consider the case in which two triangles share a vertex, it is considered self-intesecting in the shared vertex if you duplicate it. Maybe should check clipper(2) to make sure it is supported.

Btw I think this will also happen when you use CGAL's Nef Polyhedron, it is a general issue.

elalish, 2023-12-20:
I wrote a straight skeleton algorithm years ago that I've been meaning to put into C++ for Manifold - that would give us `roof()`.

kintel, 2024-07-20:
One thought: I think Manifold is stable for enough use-cases that we can move it out of experimental now. The missing piece is to add UI + cmd-line options for geometry backend selection, as we're not ready to completely ditch CGAL yet.

Also, when adding backend selection, this could be a good opportunity to clean up some of the Preferences->Advanced tab as it's getting a bit out of hand.

ochafik, 2024-07-21:
> One thought: I think Manifold is stable for enough use-cases that we can move it out of experimental now. The missing piece is to add UI + cmd-line options for geometry backend selection, as we're [...]

I'll take a look at the UI switch.

kintel, 2024-07-21:
> 
> I'll take a look at the UI switch.

I started on it, as a pretty naive first try: https://github.com/openscad/openscad/tree/kintel-geometry-engine
..but I'll be away for 3 weeks so if you want to pick this (or any subset of it) up, that would be awesome!

ochafik, 2024-07-22:
> > I'll take a look at the UI switch.
> 
> I started on it, as a pretty naive first try: https://github.com/openscad/openscad/tree/kintel-geometry-engine ..but I'll be away for 3 weeks so if you want to pick this (or any subset of it) up, that would be awesome!

https://github.com/openscad/openscad/pull/5235 may be a bit bold, I just defaulted to Manifold w/ a way to disable it. Didn't go the combo route as I'd rather we didn't offer too many equally-supported backends on the long run (e.g. don't think we there's a point in shipping fastcsg in prod), but happy to adjust course if you feel strongly against this.

kintel, 2024-07-22:
Happy to kill or bury fastcsg, but we may need to carry along CGAL + Manifold for a bit still, perhaps past next release.