openscad/openscad #4827: the text disappears

igscomp, 2023-11-13:
i have this code:

difference() {
    // Cubo
    cube([102, 25, 30]);    
    
    // Texto "Brocas de widia"
    translate([50, 0, 15])
        rotate([90, 0, 0])
            text("Brocas para pared", size = 8, valign = "center", halign = "center");
 }
 
 With F5 i can see the text, with F6, not, why???? 

:(

t-paul, 2023-11-13:
Preview is not 100% perfect. You are rotating 2D text in 3D space, which is not possible.

If you want text in 3D space, you need to `linear_extrude()` it first to convert it into a 3D object.

closed 2023-11-13 by t-paul

t-paul, 2023-11-13:
```
difference() {
    cube([102, 25, 30]);
    translate([50, 0, 15])
        rotate([90, 0, 0])
            // Convert 2D Text outline into 3D object with 2 units height
	    linear_extrude(2, center = true, convexity = 3)
	        text("Brocas para pared", size = 8, valign = "center", halign = "center");
}
```
If you add a `#` in front of the `linear_extrude()` you can see the 3D object shown in transparent red (see [Modifier_Characters](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Modifier_Characters) for more info about that).

jordanbrown0, 2023-11-13:
The issue is that `text()` does not yield a 3D object; it yields only a 2D object.    F6 render (and thus exports based on it) yield either 3D objects or 2D objects, never a mix.  The previewer expands 2D objects into one-unit-tall 3D objects, but that's just for display purposes.

You should have gotten a warning, albeit not a very clear one:
> WARNING: Scaling a 2D object with 0 - removing object 

Simpler examples like
```
cube([102, 25, 30]);    
text("Brocas para pared", size = 8, valign = "center", halign = "center");
```
yield a more clear warning:
>WARNING: Mixing 2D and 3D objects is not supported in file , line 2

igscomp, 2023-11-20:
> The issue is that `text()` does not yield a 3D object; it yields only a 2D object. F6 render (and thus exports based on it) yield either 3D objects or 2D objects, never a mix. The previewer expands [...]

Thanks for the explanation! Warning only appears first time i press F6...

igscomp, 2023-11-20:
> ```
> difference() {
>     cube([102, 25, 30]);
>     translate([50, 0, 15])
>         rotate([90, 0, 0])
>             // Convert 2D Text outline into 3D object with 2 units height
> 	    linear_extrude(2, [...]

Many thanks for the solution!!!

jordanbrown0, 2023-11-20:
>Warning only appears first time i press F6...

Yes.  Unfortunately, erroneous results are cached and there are a fair number of warnings that you don't get on subsequent runs.  (But you presumably get the erroneous results faster!)

nophead, 2023-11-20:
How easy would it be to simply not cache the result when there is an error? That way it would error again each time.

jordanbrown0, 2023-11-20:
See issue #454 for the error/caching problem.  It's spun as wanting to cache the error messages, but it could equally not cache the erroneous subassemblies.