openscad/openscad #4829: How to access "export_stl" from within my OpenSCAD script?

w1ebr, 2023-11-15:
In looking at the source files, there are several options for importing and exporting files in different formats. What is the OpenSCAD command to export the model as a STL file?

jordanbrown0, 2023-11-15:
You can't.  The language has some facilities for importing files, but no ability to write files.  (This is deliberate, so that a malicious OpenSCAD program can't scribble on your disk.)

What you need to do is either:
* From the application, F6 render (or Design/Render) then F7 export (or File/Export/Export as STL...)
* From the command line, `openscad -o <name>.stl <name>scad`.

jordanbrown0, 2023-11-15:
By the way, the mailing list is generally a better place to ask questions.
https://lists.openscad.org/list/discuss.lists.openscad.org

w1ebr, 2023-11-15:
I am processing a lot of files (34 right now). Each input STL is "import"ed. Is there a way to pass this information via command line?

w1ebr, 2023-11-15:
Also, is "manifold" an option I can try? If so, how do I do this?
Thanks!

jordanbrown0, 2023-11-15:
Better than trying to do Q&A here would be to switch to the IRC channel.  If you have an IRC client, #openscad on irc.libera.chat.  There's also a web interface at https://web.libera.chat/?channel=#openscad

jordanbrown0, 2023-11-15:
Questions answered in IRC, issue can be closed.

closed 2023-11-15 by thehans