openscad/openscad #4831: Linux 2023.11.14: Curly brackets { } won't auto indent

hamster4430, 2023-11-15:
**Describe the bug**
Using curly brackets { } doesn't put/remove an automatic indent in development builds. 

**To Reproduce**
Steps to reproduce the behavior:
With "Preferences - Editor - Indentation - Auto Indent" checked (otherwise it's right behavior), write in editor:
"{"
hit Enter and write:
"}"

**Expected behavior**
In 2021.01 build it'll put and then remove the indent, in 2023.11.14 it won't

**Environment and Version info:**
 - OS: Manjaro KDE Wayland
 - System: Intel Laptop 64-bit
 - OpenSCAD Version OpenSCAD-2023.11.14.ai16996-x86_64.AppImage

**Additional context**
The issue persists in any development build I tried for about half a year.

jordanbrown0, 2023-11-16:
+1 built from a pull from a couple of days ago on Windows MSYS2.

LinuxLover9, 2025-02-24:
I also noticed this is still an issue in 2024.11.14

zumpchke, 2025-05-11:
+1. Can we get this fixed?

kintel, 2025-05-11:
Anyone want to contribute this as a Pull Request?

t-paul, 2025-05-11:
I'm not sure it's worth it doing that at this point. QScintilla is dead, and IIRC indents are very specific to it.

closed 2026-08-27 by github-actions[bot]