openscad/openscad #4835: Add Backup files to recent file list for easy crash recovery

UBaer21, 2023-11-18:
**When crashing (OS or SCAD) backup files exist but are not offered**
After restart you have know about backup and search the folder to find the latest file.

**As many other Software a backup should be offered**
Simplest way is to add the backup files to the recent files list when written (maybe also increase recent list to 15). If everything went well the backup file is removed  and so the recent list will omit it too.
If openSCAD was closed unexpectedly the backup file exist and will be top of the recent list when opening.
Files from the Backup folder should only be opened read only to ensure they are saved as with a proper name - And deleted after saved under a new name.

**Describe alternatives you've considered**
Extra start screen dialog offering to restore backup.

jordanbrown0, 2023-11-18:
Best would be if it saved all state, and offered to restore state, so that it goes back to the same tabs, same open files, et cetera.  Don't make the user recover multiple tabs, re-associate the saved text with the right files, et cetera.

t-paul, 2023-11-18:
I really dislike that full session restore feature in anything that is not a web browser :smile:.

jordanbrown0, 2023-11-19:
Real power users just type the lost text again.