openscad/openscad #4836: Add docker builds for the Preview releases

DeflateAwning, 2023-11-20:
**Is your feature request related to a problem? Please describe.**
I wish I could pull a docker image for the preview releases.

**Describe the solution you'd like**
I want docker images for the preview releases tagged on the [Docker Hub for this project](https://hub.docker.com/r/openscad/openscad). The latest one available there is from 8 months ago.

pca006132, 2023-11-20:
I think the docker file can build the current release if you do a git update?

DeflateAwning, 2023-11-20:
It would still be nice if they were tagged and automatically built

t-paul, 2023-11-20:
I've disabled all auto-builds on docker-hub as we run into various timeouts due to limitations of the open source account.

I'll see if I can setup an auto-build pipeline on our new self-hosted CI (https://ci.openscad.org/). Push to docker-hub is already setup, so that should be quite easy to do.

t-paul, 2023-11-22:
Pipeline [dockerhub-openscad-dev-amd64](https://ci.openscad.org/teams/dockerhub/pipelines/dockerhub-openscad-dev-amd64/jobs/build/builds/4.1) is active and has successfully pushed an updated image.

This *should* now trigger every Monday at about 3:00 UTC, but that's the first timed job, so we'll have to see if that works :grinning:.

Before adding more auto-builds, we probably need to get some debian repo cache proxy installed so we are not hitting the download every time a build is running. Right now concourse only caches the container image resources, I believe.

I'll close once the automatic build is verified with the latest pipeline version...

t-paul, 2023-12-02:
Note: for now only x86-64, hopefully arm64 will follow at some point, but that needs some more work.