openscad/openscad #4838: preserving manifoldness in input and output

pca006132, 2023-11-20:
Both CGAL and Manifold requires the input to be a closed manifold. However, the current `PolySet` does not preserve manifoldness, it is just a triangle soup, like an STL file. Conversion to/from `PolySet` will lose topology information, and for certain geometries this will cause them to become non-manifold.

For example, consider the following code:
```
union() {
    cube([1, 1, 1]);
    translate([-1, -1, 0]) cube([1, 1, 1]);
}
```

[image]

The two points point in which the two cubes touch each other are duplicated in order to maintain manifoldness (at least in the manifold backend). If points with the exact same coordinates are merged, it will no longer be a manifold. When a mesh is converted into `PolySet`, it will lose the manifold information and it is hard to get it back.

#4703 is needed to make things work. I am opening this issue because I am not sure what is the current status of that PR.

pca006132, 2023-11-20:
probably related to #4672 as well... And note that this has nothing to do with precision nor defects

gsohler, 2023-11-20:
The status of this PR:

there are segfaults for few tests in windows build.
everything else was pass.

I did not have/get the possibilities to debug them.

gsohler, 2023-11-20:
> I usually enable address sanitizer for this kind of debugging, but so far
> I am unable to enable address sanitizer for openscad for some unknown
> reason. It will just abort when I start running.

Hi,

I have already tried that myself with the support/advice  from @t-paul but
I could not catch any error
Issue is that it appears to happen  only in windows MXE build.
Tests in MINGW did not succeed either
Efforts to automatically print the stack trace to stdout in openscad did
not progress very far.

If somebody else  is able to spot the location of the segault, I'd love to
see this PR green/clean

pca006132, 2023-11-20:
I can try to help but it may take some time.

kintel, 2023-11-21:
This would be very awesome to fix! The PolySet class is a bit of an eyesore, so getting it cleaned up is good. Let's continue the discussion in #4842 to determine if the proposed solution would work.

pca006132, 2023-11-21:
And we should probably persuade users to avoid using stl files and use things like 3mf instead. I think we don't have algorithms for recovering a manifold from stl files in general.

kintel, 2023-11-21:
Recovering manifolds from STL would be very awesome though. There are too many STLs in existence, without a corresponding 3mf, to hope for it to go away. ..but providing some guidance to nudge people off STL where possible makes a lot of sense.

pca006132, 2023-11-22:
it seems that CGAL does support fixing non-manifold triangular soup, see my comment in #4832.

kintel, 2024-02-26:
PolySet is now an indexed mesh and can preserve manifolds.

kintel, 2024-02-26:
Closing this - let's open more concrete issues for further manifoldness work.

closed 2024-02-26 by kintel