openscad/openscad #4839: CryptoPP_LIBRARIES not found (-DENALBE_CRYPTOPP=0)

PanPiotrAdamus, 2023-11-20:
Hi,

I am trying to build a version with python support (source http://pythonscad.org/) on Void.
All other dependencies are met (checked via ./script/check-dependencies.sh with cgal being build as external via the ./script/uni-build-dependencies.sh).

I installed CryptoPP package provided by the Void , crypto++-8.3.0_1 and crypto++-devel-8.3.0_1 however I am not able to build openscad any further.

To reproduce:
1. follow begining steps as described at http://pythonscad.org/
2. cd `build &&  cmake --log-level VERBOSE -DEXPERIMENTAL=1 -DENABLE_PYTHON=1 -DENABLE_LIBFIVE=1 -DENALBE_CRYPTOPP=0 ..`
3. error

`-- Checking for module 'cryptopp>=5.6.0'`
`--   No package 'cryptopp' found`
`CMake Error at /usr/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:230 (message):`
`  Could NOT find CryptoPP (missing: CryptoPP_LIBRARIES)`
`Call Stack (most recent call first):`
`  /usr/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)`
` cmake/Modules/FindCryptoPP.cmake:24 (find_package_handle_standard_args)`
`  CMakeLists.txt:394 (find_package)`

I cloned git repo as of yesterday 11-19-2023.
Please let me know what test/check should I do or what other info to provide.

One thing to note, I don't have python CryptoPP package installed via either pip or via Void pacakges (Void doens't provide it).

Thanks,

Piotr.

thehans, 2023-11-20:
Have you tried it with `ENABLE` spelled correctly?

PanPiotrAdamus, 2023-11-20:
Ahh, never compile on Sundays. Indeed, I have another error now, that should be easy to solve :)
I will close that ticket.

`CMake Error at /usr/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:230 (message):`
  `Could NOT find JPEG (missing: JPEG_LIBRARY JPEG_INCLUDE_DIR)`
`Call Stack (most recent call first):`
  `/usr/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)`
  `/usr/share/cmake-3.27/Modules/FindJPEG.cmake:106 (find_package_handle_standard_args)`
  `CMakeLists.txt:489 (find_package)`

PanPiotrAdamus, 2023-11-20:
Spelling issue with CMAKE command.

closed 2023-11-20 by PanPiotrAdamus