openscad/openscad #4846: SpaceNav support disabled on nightly builds

ggabbiani, 2023-11-22:
ENABLE_SPNAV option in CMakeLists.txt is enabled by default but the concrete build still depends on the detection of libspnav-dev (on Debian like systems) package on the build system.

Unfortunately the Docker image used for the build of nightly versions doesn't have this library installed, as a consequence the SpaceNav driver is not usable on nightly builds even if OpenSCAD actually supports it.

Steps to reproduce the behavior:

1. install and start spacenavd (free 3Dconnexion driver daemon), install spnavcfg package (configuration program)
2. attach a 3Dconnexion device to the PC (this is an optional step, the result can be checked just with the daemon installed and running)
3. start OpenSCAD and follow Edit --> Preferences --> Axes
4. The SpaceNav radio button in the OpenSCAD preferences cannot be enabled, nor any 3d mouse attached is usable . The baloon says that OpenSCAD has been compiled without support for SpaceNav.

**Expected behavior**
The SpaceNav radio button should be editable and the eventual 3Dconnexion mouse devise usable

**Environment and Version info :**
 - OS: Fedora Linux 37
 - System: Intel PC 64-bit
 - OpenSCAD Version: 2023.11.05.nightly (git 878335e)  installed from https://download.opensuse.org/repositories/home:/t-paul/Fedora_37/

**Additional context**
The same problem was already issued and solved for Fedora stable package (see [Red Hat Bugzilla – Bug 2242754](https://bugzilla.redhat.com/show_bug.cgi?id=2242754)) by installing the libspnav-devel package (this is the library name on Fedora systems) in the build system when producing the rpm. 

I guess that something similar should be done on the  Docker image used by your nightly build system.

rcarmo, 2024-05-28:
This also seems to apply to macOS. I am exploring ways to fix #5153 and noticed my latest nightly also has SpaceNav disabled in Sonoma

t-paul, 2024-05-28:
I don't have a Mac, so I'm not allowed to even use a VM because of how Apple *licenses* their OS.

rcarmo, 2024-05-28:
I still think you're conflating two different things, but I'm not going to split hairs. Not having the hardware is enough.

t-paul, 2024-05-28:
I'd suggest moving the topic to a new macOS specific issue then as this will for example need `libspnav` to be added to `scripts/macosx-build-dependencies.sh` as universal build.

benjaminbellamy, 2024-06-21:
I am running Ubuntu 24.04 with `spacenavd` up and running correctly, and I have the same problem with all OpenSCAD releases (stable/nightly, Deb, Snap, Flathub, AppImage):  
```
This driver was not enabled during build time and it thus non available.
```

Do we need to recompile OpenSCAD to get a SpaceMouse to work with it???

t-paul, 2024-06-21:
Flatpak nightly (beta) should work now.  I don't have a full table of what's enabled in which build, so I guess that would be useful.
At this point, it's pretty much me as build team with quite limited spare time, so if you can help getting things enabled in the official builds, things could move a bit faster. E.g. flatpak release build has the portals enabled already, but needs the client library to be added.

benjaminbellamy, 2024-06-21:
It does work indeed with the Flatpak nightly.

[image]