openscad/openscad #4848: Variable names longer than 32 characters?

Jimw338, 2023-11-22:
Would it be "simple" to increase the maximum length of variable names to more than 32 characters?

The "ObjectiveCAppleSwiftPolysyntheticLikeVariableNameThing" does have it's appeal when writing for "self-documenting code", and I've run into the 32-character limit with names like "doInterestingCenteredWindowCell_p"

As well (maybe harder), could OpenSCAD use Unicode internally for the symbol table?  (for CJK variable names)

t-paul, 2023-11-22:
Please give a specific example, I'm not aware of a limitation and that seems to work fine as expected:
```
ObjectiveCAppleSwiftPolysyntheticLikeVariableNameThing = 3;
doInterestingCenteredWindowCell_p = 5;
echo(
	ObjectiveCAppleSwiftPolysyntheticLikeVariableNameThing
	+
	doInterestingCenteredWindowCell_p
);
```
Gives: `ECHO: 8`

Unicode would be ok, likely following the specification C++ has introduced, but right now this is difficult because of the library reading the code does not natively support Unicode. Having it fully parse Unicode in a simple way explodes the internal tables, so we have to either find a new library with native Unicode support or a more clever way of restricting the allowed codepoints.

Anyway, the Unicode topic is covered by #3736.

t-paul, 2024-01-27:
Closing, no feedback.

closed 2024-01-27 by t-paul