openscad/openscad #4849: windows 10 2023.11.14 WARNING: [fast-csg] Corefinement corefinement mesh difference #4 failed

ephramcohen, 2023-11-22:
**Describe the bug**
I enabled fast-csg (only)
I render the included model, and get the following warnings 
WARNING: [fast-csg] Corefinement corefinement mesh difference #4 failed
WARNING: [fast-csg] Corefinement corefinement mesh difference #5 failed with an error: CGAL ERROR: precondition violation!
Expr: is_polygon_soup_a_polygon_mesh(polygons)
File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Polygon_mesh_processing/polygon_soup_to_polygon_mesh.h
Line: 281
Explanation: Input soup needs to define a valid polygon mesh! See is_polygon_soup_a_polygon_mesh() for further information.

**To Reproduce**
enable fast-csg and render (F5 then F6) the included code.

**Expected behavior**
I expect no warnings

**Code reproducing the issue**
difference() {
  hollowTubeOD(30, 60, 2);
    hole();
  rotate([0, 0, 36/2])
    translate([0,0,10])
      hole();
}

module hole() {
  translate([0,30,sqrt(50)])
    rotate([0,45,0])
      cube([10,10,10], center = true);
}

module hollowTubeOD(height, OD, wallThickness) {
  difference() {
    cylinder(height,d=OD);
    translate([0,0,-0.5])
      cylinder(height+1, d=OD-2*wallThickness);
  }
}

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment and Version info (please complete the following information):**
Windows 10 22H2
OpenSCAD Version 2023.11.14

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

**Additional context**
Add any other context about the problem here.

pca006132, 2023-11-23:
[image]

This is considered non-manifold, so CGAL does not accept this as a mesh.

kintel, 2024-11-28:
Closing as fast-csg has been deprecated

closed 2024-11-28 by kintel