openscad/openscad #4850: OpenSCAD loses performance in render on M2 mac when window loses focus

MWStonton, 2023-11-23:
This seems similar to issue #2054 but with much more modern OS, CPU type, OpenSCAD versions, etc. Apple are currently selling multi-tasking capabilities of macOS so logically it shouldn’t happen? Maybe the behaviour of the E cores (100% usage of 1 core by OpenSCAD without obvious progress in render) when the P cores are lost might help indicate what’s going on…

This seems to be true with latest "dev" releases (e.g. 2023.11.20) (ARM code) AND the 2021.01 release (x86 code run using Rosetta) (I've only measured the former).

For "dev" release, configured for fast-csg-safer
Unknown if issue occurs with manifold as processing is so fast

I tend to run with just the display and console windows open (and use an external editor - but this occurs without an editor active on the source file)

Device is mac Mini M2 Pro, 16GB RAM, macOS 14.1.1

If the OpenSCAD window has focus throughout, it uses 100% of a CPU core and that is always (as far as I can tell from viewing Activity Monitor) 100% of one Performance core (shared across the P cores).

If the OpenSCAD window doesn't have focus then it "randomly" switches between 100% usage of a Performance core and 100% usage of an Efficiency core (I think Activity Monitor in %'s doesn't distinguish between a Performance and Efficiency core when it says 100% so it’s 100% of A core rather than a measure of amount of processing)

With an example design: (suspect this isn’t source code dependent but will provide if requested)

WITH focus render takes approx 3m35s (+/- a few s) (time reported by OpenSCAD)

Where focus is removed (for example start render and then click on Activity Monitor) I have 3 examples of the time taken (measured with stopwatch) being 6m32s, 6m45s and 7m13s. In these examples there were 2, 4 and 3 periods where only the Efficiency cores were in use.
ex1 0-2:37=P, 2:37-4:25=E, 4:25-5:05=P, 5:05-6:32(end)=E
switch points only listed for ex2,3
ex2 0 0:56 2:14 2:55 4:01 4:53 5:09 5:55 6:45(end)
ex3 0 1:02 1:33 3:00 5:18 6:00 7:13(end)
(don’t know if it says anything that it always completes here while running on the E cores where I don’t think much (real render) processing is done)

Perhaps interesting is adding up the times spent in P core and E core processing for each example:
ex1 = 197s of P processing, 195s of E processing
ex2 = 195s of P processing, 210s of E processing
ex3 = 191s of P processing, 242s of E processing

This suggests that there's not much real OpenSCAD (render) processing going on when only the Efficiency cores are running (but it is keeping 100% of an E core busy) and trying to look a little further:

WITH focus
CPUTime (from Act. Mon.) = 3:38:54, OpenSCAD reports 3:35.290

WITHOUT focus
CPUTime                  = 7:39:29, OpenSCAD reports 7:24.728

i.e. the reported CPUTime and OpenSCAD reports are much the same also indicating that OpenSCAD isn't doing anything useful active during the added Efficiency only periods => suggests perhaps something is causing OpenSCAD to pause at "random" points for "random" periods of time but is busy waiting for something (like getting a P core back)

I believe giving focus back (when Activity Monitor indicates only E cores in use, click on the OpenSCAD window) returns OpenSCAD to normal processing with (a, 100% of 1 spread across all) Performance core(s) and you can click off OpenSCAD (to remove focus again) and normal processing will continue until another "random" point, or completion.

In case it is related, some observations:

if you remove focus very quickly from OpenSCAD after clicking the render button, the preview image remains displayed until processing is complete when it is replaced by the render image. Giving focus back to OpenSCAD will (at least at some point) clear the preview image. Instead, allowing a bit more time before removing focus at the start, the preview image is removed and then the render image is drawn at completion of processing. Focus is NOT needed for the render image to be displayed once processing is complete.

Also, the progress bar starts in blue but if focus is removed, at some point later (which may be where the P core processing is lost) turns to dark gray on a lighter gray background. Maybe the cause of both things is the same..

kintel, 2023-11-23:
Perhaps it's possible to request a performance core for the geometry processing thread

MWStonton, 2023-11-24:
Sounds logical and would be lovely if it was as simple as that but somehow I feel it’s more complex….

It could be Apple doing something to stop “background” tasks taking processing power (even if there’s very little else running) but I’d hope that wasn’t the case when I see a lot of video ads from Apple on YouTube at the moment selling the multi-tasking performance of their machines. Since only one task can have focus you can’t sell multi-tasking if you actively shut down any other tasks - so it feels like it’s something in the code, probably.

There are some more specifics I can highlight in what I initially wrote:

The fact it seems to happen with the 2021 release which is x86 code as well as the latest snapshots would mean that couldn’t be fixed directly as you suggested because x86 code can’t know it’s running on an M processor and request the type of core. It may be able to assign a priority in the system of course which is effectively what you’re suggesting rather than taking it literally.

The other thing is that despite the switching between periods of “P” processing and periods of “E” processing (allowing for limited accuracy as I was viewing the Activity Monitor manually and using a stopwatch) it seems to me that pretty much the only real computation for the OpenSCAD render occurs in the “P” processing periods - assuming the ratio between a P and E core is not massive (list of times is below). That suggests that when switching to the E processing it is waiting more than anything though I believe it is still using 100% of a core so something is going on like polling (it’s not just idle waiting or there would be very little CPU usage even of an E core). That said, it’s hard to imagine there being polling but that’s what it feels like.

ex1 = 197s of P processing, 195s of E processing
ex2 = 195s of P processing, 210s of E processing
ex3 = 191s of P processing, 242s of E processing

add to those ex4 as the case where OpenSCAD keeps focus
ex4 = 215s of P processing
In ex1 18s of “lost” P processing => 195s of E processing
In ex2 20s of “lost” P processing => 210s of E processing delta 2s from ex1 in P, 15s in E
In ex3 23s of “lost” P processing => 242s of E processing delta 4s from ex2 in P, 32s in E

ex1 P:E ratio = 18:195 = 1:10.8
ex1 to ex2 delta, P:E ratio = 1:7.5
ex2 to ex3 delta, P:E ratio = 1:8

It could be there’s always some E processing (even with 100% focus) I don’t see visually because everything swaps around too fast and I’m just looking at the visual display but it does feel rather more like idling (but busy) more than the P:E performance ratio being this high, UNLESS the clock is reduced massively during the E periods and 100% means usage at whatever the clock rate is - in which case it could be a simple “idle” wait rather than a “busy” wait for something. I’m also assuming that it wouldn’t be seriously downclocked proper render processing when it’s on the E core as that wouldn’t make sense if the demand for more performance is there.

I’m simply concluding here that even if processing was moved to an E core some of the time, the impact seems to be greater than the relative processing performance available between P and E cores…

It does seem remarkably like the issue #2054 referred to from 2017 where you commented about separate threads for the UI and processing though it feels almost like it’s one waiting for something in the other (such as when the progress bar is supposed to be updated or something like that). I just pick that as an example where there must be communication between the 2 threads - the processing thread telling the UI thread that processing is X% complete so that the UI thread can update the progress bar. There’s also the non removal of the preview image as mentioned in some cases which presumably should be done by the UI thread when the processing thread says it has started rendering. Again, I have no idea exactly what is done but it is another likely communication point between the 2 threads and potentially where the UI thread being held up causes the processing thread to be paused - if not at that point, maybe later because the next message can’t be sent from the processing thread to the UI thread.

Just my thoughts and rambling!

pca006132, 2023-11-24:
probably related: https://eclecticlight.co/2022/01/24/how-you-cant-promote-threads-on-an-m1/

I think the process is considered background task and gave a background QoS, thus running on e cores exclusively?

Also it is weird that e core processing is not doing useful computation. From my understanding fastcsg does not use multiple threads, so there is nothing to poll for... Someone can try to use a profiler to figure out why this happens.

pca006132, 2023-11-24:
Two possible things here:

1. https://doc.qt.io/qt-6/qthread.html#Priority-enum, we can change the worker thread priority so it does not inherit the GUI thread priority. I think when you lose focus on the GUI window, the QoS of the GUI thread changes, maybe the worker thread inherits this.
2. https://opensource.apple.com/source/libpthread/libpthread-218.60.3/pthread/spawn.h.auto.html, this directly changes the process QoS.

This requires some actual testing.

pca006132, 2023-11-24:
Also need some work for tbb: https://github.com/oneapi-src/oneTBB/issues/896

MWStonton, 2023-11-24:
Does the preview image sometimes not getting cleared give a clue? If the render if started and focus lost quickly (must be VERY quickly presumably) then the preview image isn’t cleared (if focus isn’t returned, until the end of processing**). I believe (and will run again tomorrow to check) that processing is continuing for some time using P cores so it must be the UI thread which has stopped in some way. If we then know that is the case then maybe it could be that there’s no problem with the processing thread until it wants to talk to the UI thread and there's something wrong with (the priority?) of the UI thread.
**another question - if it (preview image being cleared) has not happened all the way through processing what makes it happen at the end of processing - does the (presumed) completion of the processing thread then allow it to happen. Is there a further clear process as part of the draw render image at the end (what I’m asking here is whether the clear has actually occurred at all or been lost - would we know, if the "draw render image" is where the "draw preview image" would have been. Just wondering whether some of the UI stuff is just delayed or whether it has been skipped/lost somehow - doesn’t sound like computers skipping stuff unless there is a timeout involved.

Was wondering when mentioned about the processing thread inheriting the priority of the UI (which it is called/created from) …. would it keep that priority then throughout its life or follow changes in the UI priority. If it keeps it then I don't think it fits but if it follows it then that fits with the fact it (use of P cores) can be restarted by returning focus to the UI.

It does still seem strange that the E processing doesn’t seem to achieve anything. I would expect much more marginal differences between P and E processing.

Another thought re: Intel and big little - if this is the same as #2054 then it (the bug) was already there before the M1 was released and before Intel started using big little so that may well not be the issue directly and it’s down to priority in some way (just happens that among its tools on a modern device, priority can cause switching between big and little cores). i.e. if all cores are the same then we could still see the same symptoms by clock speed reduction or simply the task given less time to run it just happens that when we have big little the change is more obvious

Afraid I can't help with testing - have some knowledge of how these things work at a high level but know nothing about implementation, etc. I'll continue to poke holes in ideas if I think there might be any but if it stops being useful please let me know!

pca006132, 2023-11-24:
Can you compile openscad? If yes, I can probably try to give you some patch to test... I don't have a mac so I cannot test it.

MWStonton, 2023-11-24:
Afraid at the moment I'd have to say I can't (compile). New to mac (couple of months) and literally only run OpenSCAD and a slicer on it (use ChromeOS for everything else) and not deeply into code (and only joined github yesterday to be able to post this issue).
Would be interested to hear if you think you've found anything but understand how long posting takes so not a priority.

pca006132, 2023-11-24:
Someone who can build from source on mac can try #4853. @t-paul is there a way to let the CI output the artifact for mac so users can try the fix?

t-paul, 2023-11-24:
@pca006132 Yes, if the branch name ends in '-macos', CircleCI will also do the macOS build. Our free plan has 500 macOS minutes per month which is only about 30 builds with the current build time, so it's not enabled by default.

pca006132, 2023-11-24:
@MWStonton Here is a build that you can try: https://app.circleci.com/pipelines/github/openscad/openscad/4270/workflows/9a11f304-fa5f-419f-89cf-6dc8522336a0/jobs/17214/artifacts

MWStonton, 2023-11-24:
Just tried it and I’m afraid that as far as I can tell it’s the same.

I clicked on render and then immediately on the desktop and (as reported by OpenSCAD) the render took 6m27s (vs approx 3m30s). The first switch from P to E cores was at quite an early stage.
I also tried a render leaving focus on OpenSCAD and it completed in the normal time (3mm33s).
I noticed in both cases that the preview image was not cleared until the end of processing (when the information is available to draw the render image) (so maybe cleared late or maybe not cleared, just “drawn over” with render image?). During the render I clicked around generally in the OpenSCAD window for a while and it didn’t force the preview image to be cleared however I’ve just tried again (probably after some other processes in OpenSCAD have completed) and if I click in the OpenSCAD window the preview image is cleared (i.e. before render is complete).

I must admit to still having a gut feeling that there’s something about the UI thread rather than the processing thread or something about communication between them which is the issue i.e. why would clearing of the preview image be delayed? That’s the biggest area where anything stands out to me as being really odd behaviour!

Thanks very much for looking into this. There is an obvious workaround - to leave focus on OpenSCAD so it’s not the end of the world BUT I just wonder if I’m seeing the (multiple) long delays and wonder if there are more tiny delays (but maybe in large numbers) affecting the processing throughout.

I have an odd more minor issue with the slicer (stl to gcode “conversion” for 3D printing) I use (which is x86) which “feels” like it’s Apple GUI related - quite hard to explain but I’ll try as there’s something about it which feels a bit like this. After I slice I can do a slice preview which allows me to look at all of the steps on each printing layer and each layer. You can step up and down the layers with a slider control which works o.k. on the mac (or Windows or Linux - though I don’t think it uses HW acceleration on Linux so is processor dependent) and is reasonably fast. There’s also a text box with the layer number and up/down arrows to control which layer is displayed. You can use the arrows on the GUI or on the keyboard (when clicked in the text box) to step up and down the layers. In Windows this works really well on even a basic machine (J4005 is the example I have) but it uses 100% of the GPU and the CPU usage backs up a little and increases a bit over time while the GPU is catching up. On a faster machine, less CPU is used (and probably faster graphics acceleration) lets it complete faster. On macOS it is very slow and if you’ve held down the arrow keys for a while it clearly goes through the drawing of each layer before it completes (there can be a large backlog on releasing the arrow key that needs to be finished). On Windows it completes much quicker and it may just be it’s running faster OR it feels like it might just stop and jump to the end rather than completing all of the steps in between whereas macOS is slower and it is clear that it’s completing all steps. The slicer if FlashPrint from FlashForge. It just feels a bit like something waiting for something else to complete (processing to UI communication or CPU to GPU communication) in both OpenSCAD and FlashPrint.

MWStonton, 2023-11-24:
Just a note about my setup. Most of the time I'm accessing the mac using Chrome Remote Desktop from a Chromebook. Before the original issue report I verified that the same thing happened when using the machine directly as it does when connected via CRD. I've only tested your new code so far via CRD but will try direct use of the mac later (I'm not available for a bit so it will be this evening UK time) but will be surprised if it changes anything.
I'll assume for the moment that just having CRD installed doesn't cause the issue given that this sounds just like the 2017 issue report!

pca006132, 2023-11-24:
Ah, this looks complicated. I don't think I can fix it without spending time debugging on a Mac. I will just leave this for some Mac expert to fix for now.

Sounds unlikely that this will be related to CRD. It is possible that it inherits the QoS of the background service and we cannot increase the QoS, but considering it works fine when focused it should have high enough QoS at least on foreground.

MWStonton, 2023-11-24:
Agree - thanks for spending the time and trying

MWStonton, 2024-01-29:
A basic symptom of this which might be relatively easy to examine:

From OpenSCAD window with design loaded and displaying preview image:

Click on render and don’t click off OpenSCAD (i.e. leave focus there) and the preview image remains displayed (unlike Windows or Linux where the preview image is cleared “straight away”) until render completes and render image is displayed (writes over it) BUT if you click off OpenSCAD (loses focus) and click back then on this last operation the preview image in cleared.

(using a design which takes a while to render, enabling no “features” (to slow processing)  and which uses “color” so that the preview and render images look different makes this all easier to see visually)

While this, and the original report as the title aren’t major issues as such, I wonder if there is an impact on rendering performance which is less obviously occurring.

I was thinking that this particular symptom must be contained within a pretty small piece of code (between render being selected and expecting the preview image to be cleared) and as such be relatively easy to find (to someone that can). It feels like the code (GUI to processing thread signalling?) is one step behind where it should be and an additional “event” caused by regaining focus allows progress…

kintel, 2024-02-03:
@MWStonton I think your latest observation just reveals how refresh/redraw events are fired slightly differently on different native GUI system. Even on macOS, the behavior differs between release and debug builds.

MWStonton, 2024-02-14:
@kintel you may be right about events - I "feel" there may be more to it
@pca006132  wrote
> probably related: https://eclecticlight.co/2022/01/24/how-you-cant-promote-threads-on-an-m1/
>
> I think the process is considered background task and gave a background QoS, thus running on e cores [...]

I just re-read this and agree (think the same) about fastcsg being single threaded but in searching around the web a bit have found a few notes about macOS expecting the GUI thread to be the main thread (I don't have enough knowledge to know if that is normal or not but the suggestion was that it might not be) so "maybe" although "fastcsg processing" is a single thread, OpenSCAD could therefore be expected to be 2 thread (1 processing, 1 GUI). That could potentially explain an active thread but which is not actually achieving anything .... perhaps.
This "feels" like the kind of thing that might explain the oddities I've found with the GUI response (e.g. when the preview image is cleared after render is started). I might be adding 2 + 2 and getting 5, but that's what my gut feeling is telling me - and I'm not too concerned about what I've reported directly (the only things I can report as I can actually see them) but do wonder if there's more performance limitation (that I can't see directly) as a result of the same thing...

Just out of interest (and I know this can be true because macOS might do some things badly, or similarly Windows) but for the code I use as something of a benchmark (the text wrapping around a cylinder thing) it takes around 217s in macOS (M2 Pro Mac Mini) (with full focus) but if I install a Debian 12 ARM Virtual Machine in UTM (uses QEMU) on the same machine and run the same render using the last dev. image of OpenSCAD for ARM Linux from September 2023 it runs in around 100s. This could be just something about macOS, or macOS libraries used by OpenSCAD not being as good or something about the macOS OpenSCAD code - or a combination.
As another reference on an Intel NUC (gen 13) with Windows the render runs in 193s but in Linux (installed) it runs in 101s so it could all just be that Linux is a lot faster!