openscad/openscad #4856: OpenScad hangs

threed-factory-store, 2023-11-24:
**Describe the bug**
The OpenSCAD viewer hangs.

If I open the attached HangExample.scad in the OpenSCAD viewer, it works. But if I uncomment lines 7 and 326, the viewer hangs.  

If I run openscad-nightly and ask it to generate an STL file, it works with either version of the file.

I've added echo() calls to HangExample.scad, and they never get called in the uncommented version.

**To Reproduce**
Steps to reproduce the behavior:
1. Uncomment lines 7 and 326 in HangExample.scad
2. Open in the OpenSCAD viewer.

**Expected behavior**
Model is displayed.

**Code reproducing the issue**
Attached.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment and Version info (please complete the following information):**
 - Ubuntu 23.10
 - Dell XPS 15
 - OpenSCAD Version  2023.10.12

**Library & Graphics card information**
OpenSCAD Version: 2023.10.12
System information: Linux 6.5.0-13-generic #13-Ubuntu SMP PREEMPT_DYNAMIC Fri Nov 3 12:16:05 UTC 2023 x86_64 Ubuntu Core 20 12 CPUs 15.25 GB RAM
User Agent: OpenSCAD/2023.10.12 (Linux x86_64; Ubuntu Core 20)
Compiler: Clang "10.0.0 " 64bit
MinGW build: No
Debug build: No
Boost version: 1_71
Eigen version: 3.3.7
CGAL version, kernels: 5.5.2, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.1
Qt version: 5.12.8
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.64.6
lodepng version: 20210627
libzip version: 1.5.1
fontconfig version: 2.13.1
freetype version: 2.10.1
harfbuzz version: 2.6.4
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: fast-csg*, fast-csg-safer, fast-csg-debug, manifold, roof, input-driver-dbus, lazy-union, vertex-object-renderers, vertex-object-renderers-indexing, vertex-object-renderers-direct, vertex-object-renderers-prealloc, textmetrics, import-function, predictible-output
Application Path: /snap/openscad-nightly/3952/usr/bin
Documents Path: /home/walt/snap/openscad-nightly/3952/.local/share
User Documents Path: /home/walt/snap/openscad-nightly/3952
Resource Path: /snap/openscad-nightly/3952/usr/share/openscad-nightly
User Library Path: /home/walt/snap/openscad-nightly/3952/.local/share/OpenSCAD/libraries
User Config Path: /home/walt/snap/openscad-nightly/3952/.config/OpenSCAD
Backup Path: /home/walt/snap/openscad-nightly/3952/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
/home/walt/snap/openscad-nightly/3952/.local/share/OpenSCAD/libraries
/snap/openscad-nightly/3952/usr/share/openscad-nightly/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
/usr/share/fonts
/usr/local/share/fonts
/home/walt/snap/openscad-nightly/3952/.local/share/fonts
/home/walt/snap/openscad-nightly/3952/.fonts
/snap/openscad-nightly/3952/usr/share/fonts
[... 380 lines cut ...]
GL_NV_copy_image
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_fragment_shader_interlock
GL_NV_half_float
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays

[HangExample.zip](https://github.com/openscad/openscad/files/13461490/HangExample.zip)

**Additional context**
Add any other context about the problem here.

t-paul, 2023-11-24:
This likely hits a corner case in the `fast-csg` experimental feature you have enabled. I'd suggest disabling that.
To get a similar (usually even much faster) speedup compared to normal processing, you can try enabling the `manifold` experimental feature.

threed-factory-store, 2023-11-24:
Much better, thank you!

Manifold produced weird shapes.

closed 2023-11-24 by threed-factory-store

pca006132, 2023-11-24:
What weird shape? Is it producing incorrect output?