openscad/openscad #4859: STL import. Easy (I hope) fix for degenerated triangles.

JCU0, 2023-11-25:
Import of existing STLs is a well-known weakness of OpenSCAD. Alot of models can be 3D-printed but not imported. I have a solution for some files. I hope, is easy to understand and integrate. Also, I have a proof-of -concept Python script for STL healing.

**Geometrical details**
Let's say we have a pair of triangles: ABC and ADB. But ABC is degenerate so that C is somewhere between A and B. Lenght(AB)=Lenght(BC)+Lenght(AC) (with some epsilon as we are talking about floats). Also, triangles CBM and ACN are exist and they are valid (non-degenerade). So, model is basically valid, without holes and self-intersections. Such models are considered valid by most software.

**My solution:**
convert ABC+ADB to ADC+BCD. Same figure, same points, also same triangles count, but no degenerates!

**Python script**
Also, "proof of concept" python script is attached. No comments inside, sorry for that. Just replace "in.stl" and "out.stl" with your files and run. It will make your model OpenSCAD-compatible.

[stl_refinery.py.txt](https://github.com/openscad/openscad/files/13464958/stl_refinery.py.txt)

pca006132, 2023-11-25:
The issue is usually not this, but manifoldness issue. See https://github.com/openscad/openscad/pull/4855

JCU0, 2023-11-25:
Usual manifoldness issues can be solved with using additional software. In my case, it was Blender with 3D-printing plugin. After such reparing, model still has can't be loaded to OpenSCAD. But Blender says there is no problem, model is correct. Online STL checkers also reported 0 problems. So, I was stuck with file that is "correct" but can't be loaded by OpenSCAD.

nophead, 2023-11-25:
OpenSCAD breaks STLs on import if the vertices are closer than the snap
grid.

pca006132, 2023-11-26:
@JCU0 did you try if this works?
https://app.circleci.com/pipelines/github/openscad/openscad/4272/workflows/1c6dfd65-a89c-4b55-8168-d93d9ba56d83/jobs/17222/artifacts

Note: Need to enable manifold experimental option.

JCU0, 2023-11-27:
Tried via Wine because I have Linux.

**Before Python script:**
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation!
Expr: e_below != SHalfedge_handle()
File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_3/SNC_FM_decorator.h
Line: 418

**After Python script:**
Success

**My test model:**
https://www.thingiverse.com/thing:3351902
Life_Counter_V1.00-clip.stl
Other STLs from this thing have more serious problems and require third-party software as a first step, then py-script, then OpenSCAD.

JCU0, 2023-11-27:
Upd: Just checked that "manifold" option. It helps, model is loaded without Py script.

pca006132, 2023-11-29:
#4855 is now merged, I would recommend you to use the manifold backend and report back if you have further issues.

Closing this for now as the fix is not needed for manifold, and probably not enough for CGAL.

closed 2023-11-29 by pca006132