openscad/openscad #4860: the necessity of introducing spline curves

WinlCheng, 2023-11-27:
spline curves is a way to describe the curve.
OpenSCAD seems not to be able to make a accurate curve, this may cause some problems and reduce model quality. And more $fn cause obvious low speed of model produce.
is it possible to introduce it, or other way to describe the curve?

pca006132, 2023-11-27:
No, OpenSCAD works with mesh and is unable to deal with curves precisely. High $fn (~100) is now feasible with manifold enabled.
If you want better precision than that, you should probably look into brep kernels, e.g. OpenCASCADE.

jordanbrown0, 2023-11-27:
Hint:  Use `$fn` *only* to produce regular polygons.  Do *not* use it to control model quality.  Use `$fa` and `$fs` to control model quality.  If you use `$fn` to make your large circles good, it will make your small circles unnecessarily complex.

pca006132, 2023-12-02:
closing as there is no plan in introducing brep.

closed 2023-12-02 by pca006132