openscad/openscad #4868: Namespaces

FatherGeoffHorton, 2023-12-01:
An issue came up on the mailing list a few days ago where several libraries couldn't be used in the same program because their namespaces overlapped. Other languages (e.g., *cough cough* python) provide for namespaces to alleviate this problem. Is something similar doable in OpenSCAD?

FatherGeoffHorton, 2023-12-01:
How about calls internal to the file?

jordanbrown0, 2023-12-01:
There's nothing like a solid design here yet, but I would guess that the imported file would not have any access to the caller's variables, functions or modules, except when passed to it as parameters.

Come to think of it, $ variables would be another potential namespace conflict, though a careful client could control what gets passed to each library.

coryrc, 2025-08-25:
Duplicate of #522

coryrc, 2025-12-15:
Someone please close as duplicate of #522

closed 2025-12-15 by t-paul