openscad/openscad #4870: retire fastcsg?

pca006132, 2023-12-02:
The experimental fastcsg option uses CGAL's mesh processing when possible, which is a lot faster than CGAL's Nef Polyhedron. However, it will not work with geometries that are slightly self-intersecting (sometimes are necessary to preserve manifoldness), such as the touching cubes example:

```scad
union() {
    cube([1, 1, 1]);
    translate([-1, -1, 0]) cube([1, 1, 1]);
}
```
[image]

There are three problems:
- Robustness: It is not as robust as manifold and must fallback to Nef polyhedron for these degenerate cases. User may also turn on both fastcsg and manifold and get weird result/performance.
- Performance: There are cases in which fastcsg is slower than Nef polyhedron, and it is slower than manifold. (https://gist.github.com/ochafik/70a6b15e982b7ccd5a79ff9afd99dbcf#results)
- Maintenance issue: Half of the files in `src/geometry/cgal` are about fastcsg, retiring fastcsg may allow us to simplify a lot of the logic.

So I propose to remove fastcsg while we are also doing other refactoring. Please let me know if you got anything working with fastcsg but not manifold so I can look into it.

pca006132, 2023-12-02:
@ochafik what do you think as the author of both features?

ochafik, 2023-12-08:
Sounds good to me, my only concern would be that probably few people have been playing with manifold so far, so I’d start by enabling it by default and keep fastcsg as a backup maybe, to allow more bug reports to trickle in & be fixed before we drop the fallback. Or we say that nef3 is the backup and just see what happens 😂

https://github.com/openscad/openscad/issues/4825 has an item about considering a coexistence of the two

kintel, 2024-11-28:
fast-csg has been deprecated now.

closed 2024-11-28 by kintel