openscad/openscad #4871: Minkowski issue with Manifold enabled

t-paul, 2023-12-02:
**Describe the bug**
The example was posted on the [mailing list](https://lists.openscad.org/empathy/thread/LDHO7C6LUQ367LH2EDCSR5WLUNNFIJ5N?hash=FKKYVQ53FBXQSUMTKWC2ANZ4JXQ5ZFON#FKKYVQ53FBXQSUMTKWC2ANZ4JXQ5ZFON). It works (slow) with CGAL but produces empty result with Manifold enabled.

**To Reproduce**
```
$fa = 6;
$fs = 0.25;
inf = 1e10;      // very big
eps = 1/128;     // small fudge factor to stop CSG barfing on coincident faces.
layer_height    = 0.25;
teardrop_angle  = 45;

function r2sides(r) = $fn ? $fn : ceil(max(min(360/ $fa, r * 2 * PI / $fs), 5));    //! Replicates the OpenSCAD logic to calculate the number of sides from the radius
function r2sides4n(r) = floor((r2sides(r) + 3) / 4) * 4;                            //! Round up the number of sides to a multiple of 4 to ensure points land on all axes

module circle4n(r, d = undef) {                     //! Circle with multiple of 4 vertices
    R = is_undef(d) ? r : d / 2;
    circle(R, $fn = r2sides4n(R));
}

module extrude_if(h, center = true)                 //! Extrudes 2D object to 3D when `h` is nonzero, otherwise leaves it 2D
    if(h)
        linear_extrude(h, center = center, convexity = 5) // 3D
            children();
    else
        children();                                 // 2D

       
module teardrop(h, r, center = true, truncate = true, chamfer = 0, chamfer_both_ends = true, plus = false) { //! For making horizontal holes that don't need support material, set `truncate = false` to make traditional RepRap teardrops that don't even need bridging
    module teardrop_2d(r, truncate) {
        er = layer_height / 2 - eps;    // Extrusion edge radius
        R = plus ? r + er : r;          // Corrected radius
        offset = plus ? -er : 0;        // Offset inwards
        hull()
            for(side = [0 : 1])
                mirror([side, 0, 0])
                    intersection() {
                        hull()
                            translate([offset, 0]) {
                                circle4n(R);

                                if (teardrop_angle > 0) {
                                    x = R * tan(teardrop_angle / 2);
                                    if(truncate)
                                        translate([0, R / 2])
                                            square([2 * x, R], center = true);
                                    else
                                        polygon([[0, 0], [eps, 0], [0, R + x * tan(min(teardrop_angle, 90 - eps))]]);
                                }
                            }
                        translate([0, -2 * R])
                            square([R, 4 * R]);
                    }
    }

    render(convexity = 5)
        extrude_if(h, center)
            teardrop_2d(r, truncate);
}

module offset_3D(r, chamfer_base = false) { //! Offset 3D shape by specified radius `r`, positive or negative.
    module ball(r)
        if(chamfer_base)
            rotate_extrude()
                intersection() {
                    rotate(180)
                        teardrop(0, r);

                    translate([0, -r])
                        square([r, 2 * r]);
                }
        else
            sphere(r);

    if(r > 0)
        minkowski() {
            children();

            ball(r);
        }
    else
        if(r < 0)
            render() difference() {
                cube(inf / 2, center = true);

                minkowski() {
                    difference() {
                        cube(inf, center = true);

                        children();
                    }
                    ball(-r);
                }
            }
        else
            children();
}

module round_3D(r, ir = undef, or = undef, chamfer_base = false) { //! Round a 3D child single radius or separate inside and outside radii
    IR = is_undef(ir) ? r : ir;
    OR = is_undef(or) ? r : or;
    offset_3D(OR, chamfer_base)
        offset_3D(-OR -IR, chamfer_base)
            offset_3D(IR, chamfer_base)
                children();
}

module shape()
    difference() {
        square(40, center = true);

        square([20, 20], center = true);
    }

module rounds() {
      translate([50, 0])
        round_3D(or = 4, ir = 2, chamfer_base = true, $fn = 16)
            linear_extrude(40, center = true)
                shape();
}

rounds();
```

**Screenshots**
Expected
[image]

With Manifold enabled:
WARNING: [manifold] Minkowski failed with error, falling back to Nef operation: CGAL ERROR: precondition violation!
Expr: ! running
File: /usr/include/CGAL/Timer.h
Line: 83
[image]

**Environment and Version info (please complete the following information):**
 - OS: Debian/Testing
 - System: Dell XPS13
 - OpenSCAD Version: 2023.11.28.nightly (git 539f1f1) / OBS Build

**Library & Graphics card information**
```
OpenSCAD Version: 2023.11.28.nightly (git 539f1f1)
System information: Linux 6.5.0-4-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.5.10-1 (2023-11-03) x86_64 Debian GNU/Linux trixie/sid 8 CPUs 15.32 GB RAM
User Agent: OpenSCAD/2023.11.28.nightly (git 539f1f1) (Linux x86_64; Debian GNU/Linux trixie/sid)
Compiler: GCC "13.2.0" 64bit
MinGW build: No
Debug build: No
Boost version: 1_74
Eigen version: 3.4.0
CGAL version, kernels: 5.5.1, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.6.0
Qt version: 5.15.10
QScintilla version: 2.14.1
InputDrivers: JoystickInputDriver
GLib version: 2.78.1
lodepng version: 20210627
libzip version: 1.7.3
fontconfig version: 2.14.2
freetype version: 2.13.2
harfbuzz version: 8.0.1
cairo version: 1.18.0
lib3mf version: 1.8.1
Features: fast-csg, fast-csg-safer, fast-csg-debug, manifold*, roof, input-driver-dbus, lazy-union, vertex-object-renderers*, vertex-object-renderers-indexing*, vertex-object-renderers-direct*, vertex-object-renderers-prealloc*, textmetrics*, import-function*, predictible-output*
```

pca006132, 2023-12-02:
interesting, wonder if cgal is treating manifold output as self-intersecting.

nophead, 2023-12-02:
LOL, that is code from NopSCADlib.

I can't test it because OpenSCAD just bombs out with manifold ticked on Win11, otherwise I could run all the library tests with it.

t-paul, 2023-12-02:
@nophead yes, that's from your mailing list post from a while ago (linked above).

pca006132, 2023-12-02:
I think you are using the binary from #4834? It was something 2 weeks ago, and there was a bug in the windows build that make things crashes.

iirc @rcolyer checked your library and it works fine without the fix in #4834, so maybe you can try the snapshot from openscad.org?

t-paul, 2023-12-02:
@pca006132 The example is from @nophead but it failed on my Linux system with the recent nightly builds for Debian/Testing. Do you think it's related to the `$variable` issue? I was not expecting that due to the CGAL error being thrown which sounds more like some data structure conversion issue.

nophead, 2023-12-02:
My library might look like it works but without the $variable fix it doesn't produce the correct results if you say change $layer_height in one file of a project and not in others. When the STLS are produced from the command line all the files are used so they don't see any $variables without the fix. To stop it erroring I include the default definitions in the file that generates the STL. That works as long as none of the values are overridden from their defaults.

pca006132, 2023-12-02:
ah I am saying that to @nophead, I mean the crash with manifold enabled.

nophead, 2023-12-02:
No I don't think the empty result it is related to $variables.

pca006132, 2023-12-02:
OK I had a look at the code, it seems that the problem is the very large cubes. Manifold works with finite precision, and features that are less than 0.0001 are basically removed... So cubes of size 1e10 basically will not work.
I am also not aware of better ways of doing it if you cannot get the bounding box of an object.

nophead, 2023-12-02:
Yes I guessed that was the issue, so I ran the clip test, which is a lot simpler but also uses a big cube and that failed as well.

Is there  smaller value for inf that will work with manifold and still be much bigger than normal objects?

nophead, 2023-12-02:
Apart from the tests that involve large cubes the only other things that fail are like this.

This is what it should look kike:

[image]

This is what it looks like with manifold:

[image]

It is just a stack of polygonal holes rotated with respect to each other.

The code looks like this:
```
layer_height = 0.25;
eps = 1/128;     // small fudge factor to stop CSG barfing on coincident faces.

module extrude_if(h, center = true)                 //! Extrudes 2D object to 3D when `h` is nonzero, otherwise leaves it 2D
    if(h)
        linear_extrude(h, center = center, convexity = 5) // 3D
            children();
    else
        children();                                 // 2D

module translate_z(z)                                                               //! Shortcut for Z only translations
    translate([0, 0, z]) children();

function sides(r, n = undef) = is_undef(n) ? max(round(4 * r), 3) : n ? max(n, 3) : r2sides(r); //! Optimum number of sides for specified radius
function corrected_radius(r, n = undef)   = r / cos(180 / sides(r, n));                         //! Adjusted radius to make flats lie on the circle
function corrected_diameter(d, n = undef) = 2 * corrected_radius(d / 2 , n);                    //! Adjusted diameter to make flats lie on the circle

module poly_circle(r, sides = undef) { //! Make a circle adjusted to print the correct size
    n = sides(r, sides);
    circle(r = corrected_radius(r, n), $fn = n);
}

module poly_cylinder(r, h, center = false, sides = undef, chamfer = false, twist = 0) {//! Make a cylinder adjusted to print the correct size
    if(twist) {
        slices = ceil(h / layer_height);
        twists = min(twist + 1, slices);
        sides = sides(r, sides);
        rot = 360 / sides / twists * (twists < slices ? (1 + 1 / slices) : 1);
        if(center)
            for(side = [0, 1])
                mirror([0, 0, side])
                    poly_cylinder(r = r, h = h / 2, sides = sides, twist = twist);
        else
            render(convexity = 5)
                for(i = [0 : slices - 1])
                    translate_z(i * layer_height - eps)
                        rotate(rot * i)
                            poly_cylinder(r = r, h = layer_height + 2 * eps, sides = sides);
    }
    else
        extrude_if(h, center)
            poly_circle(r, sides);

    if(h && chamfer)
        poly_cylinder(r + layer_height, center ? layer_height * 2 : layer_height, center, sides = sides(r, sides));
}

module alt_polyhole_stl() {
    holes = [2.5, 2, 1.5];
    n = len(holes);
    size = [n * 10, 10, 10];
    difference() {
        translate([-size.x / n / 2, $preview ? 0 : -size.y / 2])
            cube($preview ? [size.x, size.y / 2, size.z] : size);

        for(i = [0 : n - 1])
            translate([i * 10, 0])
                if(i % 2)
                    translate_z(size.z)
                        poly_cylinder(r = holes[i] / 2, h = 2 * size.z, center = true, twist = i + 1);
                else
                    poly_cylinder(r = holes[i] / 2, h = size.z, center = false, twist = i + 1);
    }
}

alt_polyhole_stl();
```

pca006132, 2023-12-03:
> Yes I guessed that was the issue, so I ran the clip test, which is a lot simpler but also uses a big cube and that failed as well.
> 
> Is there a smaller value for inf that will work with manifold and still be much bigger than normal objects?

I'm afraid no, it really depends on the size of the other object you are dealing with. Probably need to change your API or wait for #4478...

I will investigate the other issue you mentioned.

pca006132, 2023-12-03:
weird, it looks completely fine on my side (but I am using the branch with kintel's smart pointer refactoring, as I am too lazy to switch the branch and recompile)
[image]

pca006132, 2023-12-03:
Tested with #4834, also works fine. I guess it is time to pull out my windows vm again...

pca006132, 2023-12-03:
@nophead it seems that BOSL2 uses projection and hull to build the bounding box, so it has the same size as the object.

nophead, 2023-12-03:
projection() is very slow though and sometimes throws a CGAL exception, so I try to avoid it. Setting inf to 1E4 seems to work, although there are some small pixel differences. When set to 1E5 I get gross errors in this test.

[image]

The  bezel is cut into sections for printing boxes larger than the print bed using clip().

With it set to 1E4 that is 10m, about the scale of a house but if somebody was modelling something like a jumbo jet would they lose fine details, like screw threads?

The alt_polyhole() works if I render it. It looks like a OpenCSG problem. I set convexity to 5, which is far too low but I seemed to get away with it before manifold. Even if I set it to `slices * sides * 2` , or even 1000 it doesn't change.

[image]

So it seems to be some odd interaction between manifold and OpenCSG, perhaps convexity not being propagated?

nophead, 2023-12-03:
This is my library info:
```
OpenSCAD Version: 2023.12.02 (git dbc52d79c)
System information: Microsoft Windows 11 (10.0.22621) x86_64 12 CPUs 15.64 GB RAM
User Agent: OpenSCAD/2023.12.02 (git dbc52d79c) (Microsoft Windows 11 (10.0.22621) x86_64)
Compiler: GCC "13.2.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_81
Eigen version: 3.4.0
CGAL version, kernels: 5.5, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Qt version: 5.15.11
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.70.2
lodepng version: 20210627
libzip version: 1.5.2
fontconfig version: 2.14.2
freetype version: 2.13.2
harfbuzz version: 7.3.0
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: fast-csg, fast-csg-safer, fast-csg-debug, manifold*, roof, input-driver-dbus, lazy-union, vertex-object-renderers*, vertex-object-renderers-indexing*, vertex-object-renderers-direct*, vertex-object-renderers-prealloc*, textmetrics, import-function, predictible-output
Application Path: C:/Program Files/OpenSCAD_new
Documents Path: C:\Users\palme\Documents
User Documents Path: C:\Users\palme\Documents
Resource Path: C:/Program Files/OpenSCAD_new
User Library Path: C:/Users/palme/Documents/OpenSCAD/libraries
User Config Path: C:\Users\palme\AppData\Local/OpenSCAD
Backup Path: C:/Users/palme/Documents/OpenSCAD/backups
OPENSCADPATH: R:\OpenSCADLibs
OpenSCAD library path:
R:/OpenSCADLibs
C:/Users/palme/Documents/OpenSCAD/libraries
C:/Program Files/OpenSCAD_new\libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
C:/Windows/fonts
C:/Users/palme/AppData/Local/Microsoft/Windows/Fonts
C:/Users/palme/.local/share/fonts
C:/usr/local/share/fonts
C:/usr/share/fonts
C:/usr/X11/lib/X11/fonts
C:/System/Library/Fonts
C:/Library/Fonts
C:/Users/palme/Library/Fonts
GLEW version: 2.1.0
OpenGL Version: 4.6.0 - Build 31.0.101.3790
GL Renderer: Intel(R) Iris(R) Xe Graphics
GL Vendor: Intel
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Qt graphics widget: QOpenGLWidget
QSurfaceFormat: RGBA(8888), depth(24), stencil(8)

GL Extensions:
GL_3DFX_texture_compression_FXT1
GL_AMD_depth_clamp_separate
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
[... 212 lines cut ...]
GL_KHR_context_flush_control
GL_KHR_debug
GL_KHR_no_error
GL_KHR_shader_subgroup
GL_KHR_shader_subgroup_arithmetic
GL_KHR_shader_subgroup_ballot
GL_KHR_shader_subgroup_basic
GL_KHR_shader_subgroup_clustered
GL_KHR_shader_subgroup_quad
GL_KHR_shader_subgroup_shuffle
GL_KHR_shader_subgroup_shuffle_relative
GL_KHR_shader_subgroup_vote
GL_KHR_texture_compression_astc_ldr
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_OVR_multiview
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_EXT_swap_control
```

pca006132, 2023-12-03:
I managed to get a different result with the rendering bug. I need to click F6 first and then F5. With VBO off, I got this:
[image]

With VBO on, I got this:
[image]

While the expected image should be
[image]

My library info:

<details>

```
QObject::startTimer: Timers can only be used with threads started with QThread
OpenSCAD Version: 1980.01.01
System information: Linux 6.5.12-xanmod1 #1-NixOS SMP PREEMPT_DYNAMIC Tue Jan  1 00:00:00 UTC 1980 x86_64 NixOS 23.11 (Tapir) 20 CPUs 31.03 GB RAM
User Agent: OpenSCAD/1980.01.01 (Linux x86_64; NixOS 23.11 (Tapir))
Compiler: Clang "14.0.6 " 64bit
MinGW build: No
Debug build: No
Boost version: 1_81
Eigen version: 3.4.0
CGAL version, kernels: 5.5.3, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.6.0
Qt version: 5.15.11
QScintilla version: 2.13.2
InputDrivers: 
GLib version: 2.78.1
lodepng version: 20210627
libzip version: 1.10.1
fontconfig version: 2.14.2
freetype version: 2.13.2
harfbuzz version: 7.3.0
cairo version: 1.18.0
lib3mf version: 2.2.0
Features: fast-csg, fast-csg-safer, fast-csg-debug, manifold, roof, input-driver-dbus, lazy-union, vertex-object-renderers, vertex-object-renderers-indexing, vertex-object-renderers-direct, vertex-object-renderers-prealloc, textmetrics, import-function, predictible-output
Application Path: /home/pca006132/code/openscad/build
Documents Path: /home/pca006132/.local/share
User Documents Path: /home/pca006132/Documents
Resource Path: /home/pca006132/code/openscad
User Library Path: /home/pca006132/.local/share/OpenSCAD/libraries
User Config Path: /home/pca006132/.config/OpenSCAD
Backup Path: /home/pca006132/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /home/pca006132/.local/share/OpenSCAD/libraries
  /home/pca006132/code/openscad/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  /nix/store/y3agyjbr00s9lqqv6jwxhmzq1p1dap9y-hack-font-3.003
  /nix/store/0llkc90l78aglqbgr6h7a37ivpi9kvkr-noto-fonts-cjk-sans-2.004
  /nix/store/d75km24zc0j1h41784gbpgsa236kbdkr-noto-fonts-cjk-serif-2.002
  /nix/store/4gl26yindhn1dm725xli4cprcac216d8-noto-fonts-23.11.1
  /nix/store/7dz7axacvry83qfpg4y3zllwqv0f2mcn-nerdfonts-3.0.2
  /nix/store/sv49p953ai506n9k2kk0zpafmgjwdzhy-font-cursor-misc-1.0.4
  /nix/store/51433k4chdlaqs7i1yh5a8mvp3hvd961-font-misc-misc-1.1.3
  /nix/store/0kgdjkmwkg9637gaq1zfm304jgj9qpsj-dejavu-fonts-2.37
  /nix/store/vlpk16dyzbc2hbmpfm8shhmr2hg3z9k4-freefont-ttf-20120503
  /nix/store/da20dkff6bggc1vmmgmaqy3rz2grhmsn-gyre-fonts-2.005
  /nix/store/czfgdbl3q2bsdhcfi10n8g44j4v4q4fi-liberation-fonts-2.1.5
  /nix/store/1cz50fywbvaq7lnwdli6113zr14qf2nd-unifont-15.1.04
  /nix/store/gmgds85s36nnskkpc9cq74rk84vfg57g-noto-fonts-emoji-2.038
  /nix/store/z8vssimvcfi4vsb3b9nhr6bfz51g0dam-home-manager-path/lib/X11/fonts
  /nix/store/z8vssimvcfi4vsb3b9nhr6bfz51g0dam-home-manager-path/share/fonts
  /etc/profiles/per-user/pca006132/lib/X11/fonts
  /home/pca006132/.local/share/fonts
  /nix/store/ws1g0fx4f28vyzcmcz2mc35776zyd45m-lndir-1.0.4/share/fonts
  /nix/store/85jldj870vzcl72yz03labc93bwvqayx-patchelf-0.15.0/share/fonts
[... 58 lines cut ...]
  /nix/store/czfgdbl3q2bsdhcfi10n8g44j4v4q4fi-liberation-fonts-2.1.5/share/fonts/truetype
  /nix/store/1cz50fywbvaq7lnwdli6113zr14qf2nd-unifont-15.1.04/share/fonts/opentype
  /nix/store/gmgds85s36nnskkpc9cq74rk84vfg57g-noto-fonts-emoji-2.038/share/fonts/noto
  /nix/store/7zkwn9c199zxc6dq28qmgrccwqry5yp6-dejavu-fonts-minimal-2.37/share/fonts/truetype
  /nix/store/0llkc90l78aglqbgr6h7a37ivpi9kvkr-noto-fonts-cjk-sans-2.004/share/fonts/opentype/noto-cjk
  /nix/store/d75km24zc0j1h41784gbpgsa236kbdkr-noto-fonts-cjk-serif-2.002/share/fonts/opentype/noto-cjk
  /nix/store/7dz7axacvry83qfpg4y3zllwqv0f2mcn-nerdfonts-3.0.2/share/fonts/truetype/NerdFonts
  /nix/store/sv49p953ai506n9k2kk0zpafmgjwdzhy-font-cursor-misc-1.0.4/lib/X11/fonts/misc
  /nix/store/51433k4chdlaqs7i1yh5a8mvp3hvd961-font-misc-misc-1.1.3/lib/X11/fonts/misc

GL context creator: EGL (new)
EGL version: 1.5
PNG generator: lodepng

GLAD version: 2.0.4
OpenGL Version: 4.6.0 NVIDIA 545.29.02
GL Renderer: NVIDIA GeForce RTX 3050 Ti Laptop GPU/PCIe/SSE2
GL Vendor: NVIDIA Corporation
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
```

</details>

nophead, 2023-12-03:
I ran my tests with inf set to 1e3 and it changed the results of the minkowski rounding test to be pixel accurate with CGAL. So even with 1e4, although the results look OK, there is some inaccuracy compared to CGAL. So anything on the scale of 1m is starting to lose precision.

PR #5282 (2024-09-20): updated manifold to latest version with double precision floating point
As described. Fixes #4871 #5211 (actually producing the expected output as CGAL) #5032 #5284.

New issues:
1. Test for #2841 is now failing due to CGAL not accepting the polyhedron output by manifold (consider it as self-intersection). But apparently the issue was never fixed, just behaves somewhat differently with manifold previously?
2. As manifold no longer depends on thrust, existing parallelization code (for minkowski) no longer works. Not sure what is the best way to deal with this, do we want to add thrust dependenc [... cut]
closed 2024-09-20 by kintel