openscad/openscad #4873: Latest openscad is bricked

HyperWinX, 2023-12-03:
**Describe the bug**
When i compile openscad from source, install it, launch app and open recent project, it just becames unusable, i cant click any buttons, cant resize window, and it just hangs.

**To Reproduce**
Follow compilation guide with "git submodule update --init --recursive"

**Expected behavior**
I thought, its supposed to work

**Code reproducing the issue**

**Screenshots**

**Environment and Version info (please complete the following information):**
 - OS: Arch Linux on Linux kernel version 6.6.3
 - System: CPU: AMD FX-8350 Eight-Core, 16GB of DDR3 RAM
 - OpenSCAD Version 2023.12.03, compiled from GitHub repo

**Library & Graphics card information**
OpenSCAD Version: 2023.12.03
System information: Linux 6.6.3-arch1-1 #1 SMP PREEMPT_DYNAMIC Wed, 29 Nov 2023 00:37:40 +0000 x86_64 Arch Linux 8 CPUs 15.37 GB RAM
User Agent: OpenSCAD/2023.12.03 (Linux x86_64; Arch Linux)
Compiler: GCC "13.2.1 20230801" 64bit
MinGW build: No
Debug build: No
Boost version: 1_83
Eigen version: 3.4.0
CGAL version, kernels: 5.6, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.1
Qt version: 5.15.11
QScintilla version: 2.14.1
InputDrivers: 
GLib version: 2.78.1
lodepng version: 20210627
libzip version: 1.10.1
fontconfig version: 2.14.2
freetype version: 2.13.2
harfbuzz version: 8.3.0
cairo version: 1.18.0
lib3mf version: 2.2.0
Features: fast-csg, fast-csg-safer, fast-csg-debug, manifold, roof, input-driver-dbus, lazy-union, vertex-object-renderers, vertex-object-renderers-indexing, vertex-object-renderers-direct, vertex-object-renderers-prealloc, textmetrics, import-function, predictible-output
Application Path: /usr/local/bin
Documents Path: /home/hyperwin/.local/share
User Documents Path: /home/hyperwin/Documents
Resource Path: /usr/local/share/openscad
User Library Path: /home/hyperwin/.local/share/OpenSCAD/libraries
User Config Path: /home/hyperwin/.config/OpenSCAD
Backup Path: /home/hyperwin/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /home/hyperwin/.local/share/OpenSCAD/libraries
  /usr/local/share/openscad/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  /home/hyperwin/.local/share/fonts
  /home/hyperwin/.local/share/flatpak/exports/share/fonts
  /var/lib/flatpak/exports/share/fonts
  /usr/local/share/fonts
  /usr/share/fonts
  /home/hyperwin/.fonts
  /usr/X11/lib/X11/fonts
  /System/Library/Fonts
  /Library/Fonts
  /home/hyperwin/Library/Fonts
  /usr/share/fonts/100dpi
  /usr/share/fonts/75dpi
  /usr/share/fonts/TTF
  /usr/share/fonts/adobe-source-code-pro
  /usr/share/fonts/cantarell
  /usr/share/fonts/encodings
  /usr/share/fonts/gnu-free
  /usr/share/fonts/liberation
  /usr/share/fonts/noto
  /usr/share/fonts/util
  /usr/share/fonts/encodings/large

GL context creator: EGL (new)
EGL version: 1.5
PNG generator: lodepng

GLEW version: 2.2.0
OpenGL Version: 3.0 Mesa 23.2.1-arch1.2
GL Renderer: AMD RS780 (DRM 2.50.0 / 6.6.3-arch1-1, LLVM 16.0.6)
GL Vendor: Mesa
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

**Additional context**
If i create new project, everything works fine

pca006132, 2023-12-03:
Are you using c36e19b? Can you try compiling it with `-DCMAKE_BUILD_TYPE=RelWithDebInfo` and use gdb to check if it triggered anything?

HyperWinX, 2023-12-05:
> How did EGL get turned on when using OpenCSG 1.5.1 ? Did you build with some custom parameters?

Nah, i followed guide

HyperWinX, 2023-12-05:
> Are you using c36e19b? Can you try compiling it with `-DCMAKE_BUILD_TYPE=RelWithDebInfo` and use gdb to check if it triggered anything?

Ill try

kintel, 2023-12-05:
It could also be helpful if you could dump the output of your cmake invocation, as something doesn't feel right.

pca006132, 2023-12-05:
Try a clean build (remove the build directory entirely), provide us your full cmake command and the output.

HyperWinX, 2023-12-05:
> It could also be helpful if you could dump the output of your cmake invocation, as something doesn't feel right.

Okay, ill send everything

HyperWinX, 2023-12-05:
[file.log](https://github.com/openscad/openscad/files/13561304/file.log)
Here is full build log without any additional flags

HyperWinX, 2023-12-05:
Uuuummm, i think, problem was caused by opening .scad file created in really old 2021 OpenSCAD version, nvm

closed 2023-12-05 by HyperWinX

pca006132, 2023-12-05:
But this should not cause any problem...

@kintel 
https://github.com/openscad/openscad/blob/de1e7d8b6152063981eb57250c3b4cac42c52f0a/CMakeLists.txt#L615-L630

I don't think this depends on OpenCSG version.