openscad/openscad #4883: VBO poor performance

pca006132, 2023-12-06:
**Describe the bug**
Initial frame for VBO rendering is much slower than the default rendering method on master.

**To Reproduce**
Steps to reproduce the behavior:
1. Get a complicated geometry with thousands of objects. (The one I tried has 5k spheres)
2. Run the following two commands and see the result:

```
$ time ./openscad ./complex.scad -o complex.png                  
Compiling design (CSG Products normalization)...
Normalized CSG tree has 5403 elements
Geometries in cache: 2
Geometry cache size in bytes: 55120
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0:00:00.277
./openscad ./complex.scad -o complex.png  0.30s user 0.23s system 91% cpu 0.577 total

$ time ./openscad --enable=vertex-object-renderers --enable=vertex-object-renderers-direct --enable=vertex-object-renderers-prealloc --enable=vertex-object-renderers-indexing ./complex.scad -o complex.png
Compiling design (CSG Products normalization)...
Normalized CSG tree has 5403 elements
Geometries in cache: 2
Geometry cache size in bytes: 55120
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0:00:03.521
./openscad --enable=vertex-object-renderers    ./complex.scad -o complex.png  3.48s user 0.41s system 101% cpu 3.845 total
```

This causes the initial render to freeze for several seconds.

**Expected behavior**
VBO renderer should have similar performance to the previous renderer.

**Cause**

Profiling with tracy gave me the following data (not necessarily accurate, I got some issue getting correct tracy results, e.g. unable to do GPU profiling):
[image]

There are over 10k calls to `createVertex`, while there are only around 400 calls to `gl_draw_triangle` when VBO is disabled. Apart from that, it seems that the hashtable used for VBO is also extremely expensive.

In case the tracy integration is helpful for your debugging, here is a patch:
```diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index d9883bd95..ac0cf7fe6 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -54,6 +54,7 @@ option(ENABLE_TBB "Enable support for oneAPI Threading Building Blocks." ON)
 option(ENABLE_CGAL "Enable CGAL." ON)
 option(ALLOW_BUNDLED_HIDAPI "Allow usage of bundled HIDAPI library (Windows only)." OFF)
 option(ENABLE_PYTHON "Enable experimental Python Interpreter" OFF)
+option(TRACY_ENABLE "Enable Tracy profiler support" OFF)
 include(CMakeDependentOption)
 cmake_dependent_option(APPLE_UNIX "Build OpenSCAD in Unix mode in MacOS X instead of an Apple Bundle" OFF "APPLE" OFF)
 cmake_dependent_option(ENABLE_QTDBUS "Enable DBus input driver for Qt5." ON "NOT HEADLESS" OFF)
@@ -373,6 +374,19 @@ if(ENABLE_CGAL)
   set(CMAKE_MODULE_PATH ${ORIGINAL_CMAKE_MODULE_PATH})
 endif(ENABLE_CGAL)
 
+if (TRACY_ENABLE)
+  include(FetchContent)
+  FetchContent_Declare(tracy
+    GIT_REPOSITORY https://github.com/wolfpld/tracy.git
+    GIT_TAG v0.10
+    GIT_SHALLOW TRUE
+    GIT_PROGRESS TRUE
+  )
+  FetchContent_MakeAvailable(tracy)
+  option(TRACY_NO_CODE_TRANSFER ON)
+  target_link_libraries(OpenSCAD INTERFACE TracyClient)
+endif()
+
 find_package(LibZip REQUIRED QUIET)
 message(STATUS "libzip: ${LIBZIP_VERSION}")
 target_include_directories(OpenSCAD SYSTEM PRIVATE ${LIBZIP_INCLUDE_DIR_ZIP} ${LIBZIP_INCLUDE_DIR_ZIPCONF})
@@ -1293,3 +1307,7 @@ add_sanitizers(OpenSCAD)
 if(INFO)
   include("cmake/Modules/info.cmake")
 endif()
+
+# for nixos
+set(CMAKE_CXX_STANDARD_INCLUDE_DIRECTORIES 
+    ${CMAKE_CXX_IMPLICIT_INCLUDE_DIRECTORIES})
diff --git a/src/glview/GLView.cc b/src/glview/GLView.cc
index 645d2e955..5be0961b0 100644
--- a/src/glview/GLView.cc
+++ b/src/glview/GLView.cc
@@ -7,6 +7,12 @@
 #include "degree_trig.h"
 #include <cmath>
 #include <cstdio>
+#if __has_include(<tracy/Tracy.hpp>)
+#include <tracy/Tracy.hpp>
+#else
+#define FrameMarkStart(x)
+#define FrameMarkEnd(x)
+#endif
 
 #ifdef ENABLE_OPENCSG
 #include <opencsg.h>
@@ -166,10 +172,11 @@ void GLView::paintGL()
     // FIXME: This belongs in the OpenCSG renderer, but it doesn't know about this ID yet
     OpenCSG::setContext(this->opencsg_id);
 #endif
+    FrameMarkStart("Renderer Draw");
     this->renderer->prepare(showfaces, showedges);
     this->renderer->draw(showfaces, showedges);
+    FrameMarkEnd("Renderer Draw");
   }
-
   glDisable(GL_LIGHTING);
   if (showaxes) GLView::showSmallaxes(axescolor);
 }
```

pca006132, 2023-12-06:
@kintel I also tested with #4789, same performance.

kintel, 2023-12-29:
Benchmarked this across a handful of representative models. The conclusion is that using just `--enable=vertex-object-renderers` can be very bad for performance. The recommendation is to always enable all three `--enable=vertex-object-renderers --enable=vertex-object-renderers-direct --enable=vertex-object-renderers-prealloc` if you want VBO support.

kintel, 2023-12-29:
Will be addressed by #4787

PR #4787 (2023-12-31): VBO: Make vertex-object-renderers-direct and -prealloc enabled by default
This enables the `vertex-object-renderers-direct` and `vertex-object-renderers-prealloc` functionality together with `vertex-object-renderers` is enabled, and removes them as options.

The main reason is that `glBufferSubData()` is too slow on macOS, so we'll prefer preallocating buffers, but this has proven to be very slow on Linux too for certain types of models.

Related to https://github.com/openscad/openscad/issues/4782

Fixes #4883, #4504
closed 2023-12-31 by kintel