openscad/openscad #4885: Customised mumeric strings no longer evaluating as strings.

rw-donaldp, 2023-12-07:
Spotted when trying 3rd party code from Thingiverse. I'm not sure if this is a delibarate change or a bug?
A string in the customiser combo box that consists of just numbers is no longer being processed as a string type but is being converted into a number type.

openSCAD version 2021.01 behaves differently to the the latest dev versions e.g. 2023.12.06.
```
param="A";  // [A:Letter A, 1: Number 1]

param2=(param=="A")? 99:
       (param=="1")? 1:
       50;
       
if (param=="A") {
  echo("Is's A");
} else if (param=="1") {
  echo("Is's a string 1");
} else if (param==1) {
  echo("Is's a number 1");
} else {
  echo("It's niether");
}      

echo(param,param2);
```

Version 2021.01 gives:
ECHO: "Is's a string 1"
ECHO: "1", 1
Version 2023.12.06 gives
ECHO: "Is's a number 1"
ECHO: 1, 50

jordanbrown0, 2023-12-07:
Duplicated in 2023.09.25 (on Windows, not that that is likely to matter).

Sure looks like a bug to me, even independent of the incompatibility.  The type assigned should control.

jordanbrown0, 2023-12-07:
Minimum test case:
```
param="A";  // [A:Letter A, 1: Number 1]
echo(param);
```
will produce
```
ECHO: 1
```
when it should produce
```
ECHO: "1"
```

t-paul, 2023-12-07:
We probably should follow the Thingiverse Customizer here, if that always uses strings here, we should stick to that.
I'm pretty sure Thingivers Customer does not even allow quoted strings in the options, but I have to double check again.

jordanbrown0, 2023-12-08:
Here's a useful test case:
```
number = 0;  // [1: one, 2: two, three: three]
string = "a";    // [1: one, 2: two]

echo(number=number, string=string);
```
Try it in 2021.01.  `number` can be set to either 1 or 2; attempting to select "three" resets it back to the default because of the type mismatch.  "string" can be set to either "1" or "2".

This changed in Customizer rework #3755.

If I'm reading it all right...

[Before that](https://github.com/openscad/openscad/blame/b374ae74e25a87ce442508b9b314b03b7e67ed2d/src/parameter/parameterobject.cpp#L99), the result type was driven by the original type, with the selected value coerced to string or bool, or reset if it didn't match the original type.

[The current implementation](https://github.com/openscad/openscad/blob/44b9f8f160baeead1c9e6dd7f76b0b952df1e603/src/core/customizer/ParameterObject.cc#L261), however, prefers double without regard to the original type.

I suspect that the answer is that the [creation of the `EnumParameter`](https://github.com/openscad/openscad/blob/44b9f8f160baeead1c9e6dd7f76b0b952df1e603/src/core/customizer/ParameterObject.cc#L484) should squirrel away the original type and use it to determine how to interpret the enumerated value.

Notes:
* The "new" way is inconsistent; it can be used to create an enum list that will replace a string with a number or vice versa, but cannot replace a bool with a different type, or other variations.
* The "old" way would handle a word that looked like a number by converting it to a double, and then converting it back to a string, which means that leading and trailing zeroes are dropped, et cetera.  I don't know whether a straightforward fix to this "don't change types" bug would fix this bug.

UBaer21, 2023-12-10:
I just got a file from 2017 that failed with 2021 using 
`part = "helicone_a"; // [helicone_a, -2, -1, 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18]`
where numbers need to be numbers not strings