openscad/openscad #4886: Ctrl-C in the editor deletes selection

butcherg, 2023-12-07:
**Description:**  
In the editor, _sometimes_ selecting a segment of text and pressing Ctrl will delete the selected text.  It's not in the clipboard, but it's available for restoration by Undo.

**To Reproduce:**
1. Select some text
2. Press and hold Ctrl

**Expected behavior:**
Text disappears.  Typing Ctrl-C does not restore it, but Undo does.  Does not consistently happen; may be associated with window focus.

**Code reproducing the issue**
If applicable, add trimmed down, self-contained code demonstrating the issue.

**Environment and Version info (please complete the following information):**
 - OS: Ubuntu 22.04
 - System: AMD Ryzen 5
 - OpenSCAD Version:

**Library & Graphics card information**
QObject::startTimer: Timers can only be used with threads started with QThread
OpenSCAD Version: 2023.12.05
System information: Linux 6.2.0-36-generic #37~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Oct  9 15:34:04 UTC 2 x86_64 Ubuntu 22.04.3 LTS 24 CPUs 15.53 GB RAM
User Agent: OpenSCAD/2023.12.05 (Linux x86_64; Ubuntu 22.04.3 LTS)
Compiler: GCC "11.4.0" 64bit
MinGW build: No
Debug build: No
Boost version: 1_74
Eigen version: 3.4.0
CGAL version, kernels: 5.4, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.15.3
QScintilla version: 2.11.6
InputDrivers: 
GLib version: 2.72.4
lodepng version: 20210627
libzip version: 1.7.3
fontconfig version: 2.13.1
freetype version: 2.11.1
harfbuzz version: 2.7.4
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: fast-csg, fast-csg-safer, fast-csg-debug, manifold, roof, input-driver-dbus, lazy-union, vertex-object-renderers, vertex-object-renderers-indexing, vertex-object-renderers-direct, vertex-object-renderers-prealloc, textmetrics, import-function, predictible-output
Application Path: /home/glenn/3DPrinting/openscad/build-release
Documents Path: /home/glenn/.local/share
User Documents Path: /home/glenn/Documents
Resource Path: /home/glenn/3DPrinting/openscad
User Library Path: /home/glenn/.local/share/OpenSCAD/libraries
User Config Path: /home/glenn/.config/OpenSCAD
Backup Path: /home/glenn/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /home/glenn/.local/share/OpenSCAD/libraries
  /home/glenn/3DPrinting/openscad/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  /usr/share/fonts
  /usr/local/share/fonts
  /home/glenn/.local/share/fonts
  /home/glenn/.fonts
  /usr/X11/lib/X11/fonts
  /System/Library/Fonts
  /Library/Fonts
  /home/glenn/Library/Fonts
  /usr/share/fonts/X11
  /usr/share/fonts/cMap
  /usr/share/fonts/cmap
  /usr/share/fonts/opentype
  /usr/share/fonts/truetype
  /usr/share/fonts/type1
  /usr/share/fonts/X11/Type1
  /usr/share/fonts/X11/encodings
  /usr/share/fonts/X11/misc
  /usr/share/fonts/X11/util
  /usr/share/fonts/cmap/adobe-cns1
  /usr/share/fonts/cmap/adobe-gb1
  /usr/share/fonts/cmap/adobe-japan1
  /usr/share/fonts/cmap/adobe-japan2
  /usr/share/fonts/cmap/adobe-korea1
  /usr/share/fonts/opentype/malayalam
  /usr/share/fonts/opentype/noto
  /usr/share/fonts/opentype/urw-base35
  /usr/share/fonts/truetype/Gargi
  /usr/share/fonts/truetype/Gubbi
  /usr/share/fonts/truetype/Nakula
  /usr/share/fonts/truetype/Navilu
  /usr/share/fonts/truetype/Sahadeva
  /usr/share/fonts/truetype/Sarai
  /usr/share/fonts/truetype/abyssinica
  /usr/share/fonts/truetype/arphic
  /usr/share/fonts/truetype/dejavu
  /usr/share/fonts/truetype/droid
  /usr/share/fonts/truetype/fonts-beng-extra
  /usr/share/fonts/truetype/fonts-deva-extra
  /usr/share/fonts/truetype/fonts-gujr-extra
  /usr/share/fonts/truetype/fonts-guru-extra
  /usr/share/fonts/truetype/fonts-kalapi
  /usr/share/fonts/truetype/fonts-orya-extra
  /usr/share/fonts/truetype/fonts-telu-extra
  /usr/share/fonts/truetype/fonts-yrsa-rasa
  /usr/share/fonts/truetype/freefont
  /usr/share/fonts/truetype/inconsolata
  /usr/share/fonts/truetype/kacst
  /usr/share/fonts/truetype/kacst-one
  /usr/share/fonts/truetype/lao
  /usr/share/fonts/truetype/lato
  /usr/share/fonts/truetype/liberation
  /usr/share/fonts/truetype/liberation2
  /usr/share/fonts/truetype/libreoffice
  /usr/share/fonts/truetype/lohit-assamese
  /usr/share/fonts/truetype/lohit-bengali
  /usr/share/fonts/truetype/lohit-devanagari
  /usr/share/fonts/truetype/lohit-gujarati
  /usr/share/fonts/truetype/lohit-kannada
  /usr/share/fonts/truetype/lohit-malayalam
  /usr/share/fonts/truetype/lohit-oriya
  /usr/share/fonts/truetype/lohit-punjabi
  /usr/share/fonts/truetype/lohit-tamil
  /usr/share/fonts/truetype/lohit-tamil-classical
  /usr/share/fonts/truetype/lohit-telugu
  /usr/share/fonts/truetype/lyx
  /usr/share/fonts/truetype/malayalam
  /usr/share/fonts/truetype/msttcorefonts
  /usr/share/fonts/truetype/noto
  /usr/share/fonts/truetype/padauk
  /usr/share/fonts/truetype/pagul
  /usr/share/fonts/truetype/samyak
  /usr/share/fonts/truetype/samyak-fonts
  /usr/share/fonts/truetype/sinhala
  /usr/share/fonts/truetype/teluguvijayam
  /usr/share/fonts/truetype/tibetan-machine
  /usr/share/fonts/truetype/tlwg
  /usr/share/fonts/truetype/ttf-bitstream-vera
  /usr/share/fonts/truetype/ttf-khmeros-core
  /usr/share/fonts/truetype/ubuntu
  /usr/share/fonts/type1/gsfonts
  /usr/share/fonts/type1/urw-base35
  /usr/share/fonts/X11/encodings/large

GL context creator: GLX (new)
GLX version: 1.4
PNG generator: lodepng

GLEW version: 2.2.0
OpenGL Version: 4.6.0 NVIDIA 525.147.05
GL Renderer: NVIDIA GeForce GTX 1660 Ti/PCIe/SSE2
GL Vendor: NVIDIA Corporation
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

pca006132, 2023-12-09:
I think one way to debug this kind of issue would be to use rr to record a trace that triggered the behavior, and attach it here so the devs can reproduce it easily.

bjohas, 2023-12-09:
I'm also experiencing this or something like it (OS: Ubuntu 22.04, Intel). However, I am also experiencing that just selecting text (e.g., trackpad double click) can erase text. The text is typically not restored by using undo.

t-paul, 2023-12-09:
I have seen that issue a while ago on random occations but not lately (running Debian/Testing), there is a chance this is fixed in QScintilla (the editor component).

butcherg, 2023-12-10:
No "C" in any of my occurrences, just empty space...

t-paul, 2023-12-10:
As it was only spurious it's difficult to really see what's going on. I did have the feeling that the selection removal happened already when pressing the CTRL key, so it was wrongly interpreted as input that is supposed to remove the selected text.

t-paul, 2025-04-05:
@etkaar please add which build that is (e.g. downloaded from where) and the version of QScintilla shown in LibraryInfo (like in the initial post it says: `QScintilla version: 2.11.6`).

t-paul, 2025-04-05:
Thanks, looks like we need to find a way to get an updated QScintilla into the AppImage build.

closed 2025-04-06 by t-paul

t-paul, 2025-04-06:
The next AppImage build should pick up the new editor version. The library info dialog should show
```
Qt version: 5.15.3
QScintilla version: 2.14.1
```