openscad/openscad #4893: Artifacts when rendering small project with a lot of vertices

HyperWinX, 2023-12-10:
**Describe the bug**
I tried rendering small project with a lot of vertices, and got this -_-
[image]

**To Reproduce**
Render this project with $fn set to 600

**Expected behavior**
I thought its supposed to render
I can send what its actually supposed to render
[image]

**Code reproducing the issue**
 ```
$fn = 600;

module shape() {
    render() difference() {
        sphere();
        cylinder(r=0.3, h=3, center=true);
        rotate([90, 0, 0]) cylinder(r=0.3, h=3, center=true);
        rotate([0, 90, 0]) cylinder(r=0.3, h=3, center=true);
        rotate([45, 0, 0]) cylinder(r=0.3, h=3, center=true);
        rotate([-45, 0, 0]) cylinder(r=0.3, h=3, center=true);
        rotate([0, 45, 0]) cylinder(r=0.3, h=3, center=true);
        rotate([0, -45, 0]) cylinder(r=0.3, h=3, center=true);
        rotate([90, 0, 45]) cylinder(r=0.3, h=3, center=true);
        rotate([90, 0, -45]) cylinder(r=0.3, h=3, center=true);
    }
}

module line(x, y) {
    translate([x * 2, y, 0]) shape();
    if (x > 0) line(x - 1, y);
}    

module grid(x, y) {
    line(x, y * 2);
    if (y > 0) grid(x, y - 1);
}

grid(3, 3);
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment and Version info (please complete the following information):**
 - OS: Arch Linux on kernel 6.6.5
 - System: AMD FX-8350 Eight-Core 3.7GHz, 16GB DDR3 1600MHz
 - OpenSCAD Version 2023.12.05, compiler from source

**Library & Graphics card information**
OpenSCAD Version: 2023.12.05
System information: Linux 6.6.5-arch1-1 #1 SMP PREEMPT_DYNAMIC Fri, 08 Dec 2023 17:06:25 +0000 x86_64 Arch Linux 8 CPUs 15.37 GB RAM
User Agent: OpenSCAD/2023.12.05 (Linux x86_64; Arch Linux)
Compiler: GCC "13.2.1 20230801" 64bit
MinGW build: No
Debug build: No
Boost version: 1_83
Eigen version: 3.4.0
CGAL version, kernels: 5.6, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.1
Qt version: 5.15.11
QScintilla version: 2.14.1
InputDrivers: 
GLib version: 2.78.1
lodepng version: 20210627
libzip version: 1.10.1
fontconfig version: 2.14.2
freetype version: 2.13.2
harfbuzz version: 8.3.0
cairo version: 1.18.0
lib3mf version: 2.2.0
Features: fast-csg, fast-csg-safer, fast-csg-debug, manifold, roof, input-driver-dbus, lazy-union, vertex-object-renderers, vertex-object-renderers-indexing, vertex-object-renderers-direct, vertex-object-renderers-prealloc, textmetrics, import-function, predictible-output
Application Path: /usr/local/bin
Documents Path: /home/hyperwin/.local/share
User Documents Path: /home/hyperwin/Documents
Resource Path: /usr/local/share/openscad
User Library Path: /home/hyperwin/.local/share/OpenSCAD/libraries
User Config Path: /home/hyperwin/.config/OpenSCAD
Backup Path: /home/hyperwin/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /home/hyperwin/.local/share/OpenSCAD/libraries
  /usr/local/share/openscad/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  /home/hyperwin/.local/share/fonts
  /home/hyperwin/.local/share/flatpak/exports/share/fonts
  /var/lib/flatpak/exports/share/fonts
  /usr/local/share/fonts
  /usr/share/fonts
  /home/hyperwin/.fonts
  /usr/X11/lib/X11/fonts
  /System/Library/Fonts
  /Library/Fonts
  /home/hyperwin/Library/Fonts
  /usr/share/fonts/100dpi
  /usr/share/fonts/75dpi
  /usr/share/fonts/TTF
  /usr/share/fonts/adobe-source-code-pro
  /usr/share/fonts/cantarell
  /usr/share/fonts/encodings
  /usr/share/fonts/gnu-free
  /usr/share/fonts/gsfonts
  /usr/share/fonts/liberation
  /usr/share/fonts/noto
  /usr/share/fonts/util
  /usr/share/fonts/encodings/large

GL context creator: EGL (new)
EGL version: 1.5
PNG generator: lodepng

GLEW version: 2.2.0
OpenGL Version: 3.0 Mesa 23.2.1-arch1.2
GL Renderer: AMD RS780 (DRM 2.50.0 / 6.6.5-arch1-1, LLVM 16.0.6)
GL Vendor: Mesa
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
**Additional context**
Add any other context about the problem here.

UBaer21, 2023-12-10:
works with v2023.11.23 (manifold activated) 
[image]

not sure if the recursion causes issues - you can just use ` for (x=[0:2],y=[0:2])translate([x,y]*2) shape();`
also using $fs is preferred to $fn

HyperWinX, 2023-12-10:
> works with v2023.11.23 (manifold activated) [...]

Okay, i'm rendering it right now. I have all features activated. But i need to set $fn, because if i dont, ill get only 5 vertices

UBaer21, 2023-12-10:
probably you also need to set $fa to 1(or smaller).  The idea of $fs and $fa is to have the sphere and cylinder with different resolutions as these parts have different size. Using $fn=600 for both will cause much smaller faces on the cylinders.
Also 600 is a lot for such small objects and far beyond a printer can accomplish.
a sphere(r=1,$fn=600); will have faces smaller 0.01mm (10μm) - (if exported as 3mf units will be mm)

pca006132, 2023-12-10:
Do not activate manifold alongside with fastcsg, there is no benefit in doing that.

HyperWinX, 2023-12-10:
> Do not activate manifold alongside with fastcsg, there is no benefit in doing that.

Okay, thanks

HyperWinX, 2023-12-10:
> probably you also need to set $fa to 1(or smaller). The idea of $fs and $fa is to have the sphere and cylinder with different resolutions as these parts have different size. Using $fn=600 for both [...]

Im doing that just for fun, not for printer
Anyways, after i fixed line() function it rendered fine, thanks bro

closed 2023-12-10 by HyperWinX