openscad/openscad #4894: Optional split editor pane

Alex-Kent, 2023-12-10:
Overview:
Having the ability to split a text editor pane into two independent parts lets one easily switch between code in two (possibly distant) locations in the file.  This makes editing physically-separated but related code easier.

Example use case 1:
OpenSCAD keeps user-settable parameters at the top of the file.  With complex designs one will often be working on code far down in the file.  Adding/Modifying related parameter definitions involves manually scrolling through the file, having to re-find one's place each time.

Example use case 2:
One is working on two modules that define two parts designed to work together.  When tuning their fit, etc. one often needs to make iterative changes to each module's code.  Having this feature would make editing them easier and let one see both modules' code at the same time.

The requested feature:
Add the ability to split the editor pane into two independently-scrollable sections.  Both sections should be editable.

Alternatives:
None that I can think of within the confines of the built-in editor.

UBaer21, 2023-12-10:
multiple editor windows sound nice. 
However you always can use an external editor and the "undocked" windows feature.
Also "alt+F2" / F2  to set and jump bookmarks can help navigating large code.

Alex-Kent, 2023-12-15:
Yea, that would normally be an good option.  On my system though `Control`+`F2`¹ is a global shortcut that switches virtual desktops.  This means that, for me, setting/clearing bookmarks must thus be done via the menubar which is rather cumbersome.  (I understand that customizable shortcuts is a work-in-progress.)

¹ You mentioned `Alt`+`F2` but I assume that's a typo.  On my system that shortcut brings up the desktop's "Run command" dialog.

Alex-Kent, 2024-01-28:
I ended up working around this by adding a window manager input action that remaps F1 to Alt+F2 for the OpenSCAD window.