openscad/openscad #4897: little bug with french version on version 2023.12.11 (git 3d094bf87)

terpene33, 2023-12-14:
**Describe the bug**
A clear and concise description of what the bug is.
in the interface there is some command in French Language but other in turckish language
Yeni dosia
Yeni penserede Aç  and other command.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '.fichiers..'

**Environment and Version info (please complete the following information):**
 - OS: [macOS 12 ventura, ]
 - System: [ Intel PC 64-bit]
 - OpenSCAD Version [2023.12-11.] 

** Can I help You ???  For traduction ?

terpene33, 2023-12-22:
Perhaps do to an old MacOS version ? macOS 13.6.1 (22G313)  not 14 Mojave

kintel, 2023-12-22:
On my macOS 13 laptop I don't see the Turkish strings. However, the strings you mention are not French either, they're English. Do you happen to have Turkish as a configured language in System Settings -> Language & Region?

I also notice that the strings in question look a bit odd in the translation files:
https://github.com/openscad/openscad/blob/master/locale/fr.po#L477

They have a `#, fuzzy` comment, which may confuse the lookup.

terpene33, 2023-12-22:
Yes ! it's right : there is turkish in configured langage  !
I don't use this !!!
So now there some item in English language (it's well for me and I have Add English language) or in french )
Thanks You ! You are right ! Can you write on git this issue is solved ?

My appologies for my poor English language !

Best regards ! ( my SMTP  don't work with your IP adresse ?? !?)

thehans, 2023-12-22:
> Can you write on git this issue is solved ?

Closing per request

closed 2023-12-22 by thehans