openscad/openscad #4900: We NEED a manual Dark Mode option, regardless of OS setting, please!

Nikorasu, 2023-12-19:
I've been trying to solve this issue for a couple years now, with no luck..
OpenSCAD's window-border color does not respect the OS's dark mode setting in Linux Mint 20. And there's no settings anywhere either in my OS or in the program, to change it.
The last version that did, was 2019.05, all newer Snapshots ignore it or expect some different OS flags or something.
Here's what the old version looked like:
[image]
And here's what the newer snapshot appimages look like:
[image]
Clearly those white window-borders aren't how it should look, when everything else on my system runs in dark mode. Very hard on the eyes..

Preferably, the best solution for this problem would be to provide the user with an option to set Dark Mode on/off manually, in the preferences with the other theme options. I've seen quite a few other people with similar problems, and this solution might fix it for all of us.

RajHid, 2024-04-07:
Similar problem here, OpenSCAD do not change the background color to darkmode when applied in the preferences/Editor --> Color syntax highlighting

autotoxicus, 2025-02-02:
I had this issue under Linux Mint Cinnamon from version 22.0. I found a working solution to get is all dark again.  
Keyword "Qt-Theme".

https://www.reddit.com/r/linuxmint/comments/1ec382p/comment/lexjjih/

xylr117z4, 2025-10-17:
Why wouldn't dark/light mode just be a setting in preferences?  We already have settings for the Editor and Preview.  This should absolutely just be an option.  Relying an OS settings seems rather trusting that the OS will communicate those. 

Anyway +1 for adding this as an option for all OSs, since it doesn't work on Windows either on the latest dev build (2025.10.16.)

P.S: you can force dark mode in windows by adding "-platform windows:darkmode=2" to the target of a shortcut.

TheItroma, 2025-11-10:
Also an Issue for me on arch linux
@jordanbrown0 bro, its 1 toggle, I know they compound but it would be very useful and probably MUCH easier to deal with in case that the os doesn't give the proper theming or whatever. Which seems to be an issue as you said 🤷

TheItroma, 2025-11-10:
Regardless, theres absolutely nothing wrong with a simple toggle in case the os does give wrong info

jordanbrown0, 2025-11-11:
OK, so I fired up a Linux VM (Ubuntu, gotta start somewhere), and put it into dark mode, built a current OpenSCAD, and OpenSCAD doesn't seem to notice the dark mode setting at all.

But that raises a question:  in the problem images shown above, is that a default OpenSCAD configuration, or has it been locally switched to non-default colors to get the editor and 3D view to be dark?

dieasapjordan, 2026-03-09:
> Should this be a local setting? Well, how often does somebody want OpenSCAD in light mode and the rest of their system in dark mode, or vice versa? My guess is that approximately zero people want that, and so it should **not** be a local option, cluttering the UI and offering opportunities for misconfiguration.
> 
> By all means we should have an issue about getting that one display element wrong, but we SHOULD NOT add a manual option.

I made a github account just for this. This retarded comment made me extremely angry. I have tried to get openscad on dark mode across 4 different OS's and on all of them I'm unable to get rid of the white borders.

You clearly cannot decipher OS theme configurations for this program, or have an extremely hard time doing so.
 JUST ADD A LIGHT THEME/DARK THEME TOGGLE OPTION on the preferences menu.

PEOPLE WANT A DARK THEME. I can't get my friends on openscad because they won't tolerate the white borders.
If the point is to make the theme configuration easier, it is far easier to click a toggle in a menu than to have to mess with QTwhatever desktop theme manager options.

This old fart being against the obvious and straightforward solution to this problem is exasperating.

Also how was it not obvious to you that the original poster of this issue manually changed the editor and viewer to a dark theme as a way to demonstrate the problem.

3 years and still no button. I'm not a programmer but how hard can it be to add the toggle that changes the whites to black and viceversa?

jordanbrown0, 2026-03-10:
(I think I might have manually overridden the window-frame color, so it doesn't track dark/light.  That's a Windows setting, not an OpenSCAD setting.)

t-paul, 2026-03-10:
> * Defaults for editor and 3D view.

I think that could read: Connect the dark/light schemes so it can switch between Tomorrow <-> Tomorrow Night automatically.

jordanbrown0, 2026-03-15:
I have to back off on something I said earlier.  I said that the screen shots showed OpenSCAD picking up the system's setting, and then mishandling it.  But, looking at it again, that is **not** shown by those original screen shots.  I think I was fooled by the (explicitly-set) editor and 3D view color settings, and the fact that my experiments **did** yield problems where we mishandle the setting.

Sigh.  So we have two, and maybe three, issues:
* For some platforms, Qt6 doesn't seem to be picking up dark mode. (But Qt5 did.)
* For all platforms, dynamically switching to dark mode (while the application is running) is mishandled; some UI elements are updated while others are not.
* Maybe a separate issue:  for Windows 10, it seems that we properly detect the mode (because icons switch to being white), but the background doesn't get set (and so the icons are white-on-white).  Maybe this is tied to the dynamic-switch issue, and maybe not.

I've found the causes of the dynamic-switch issue.
* We explicitly choose an icon set at startup based on whether we're in dark mode.  We do not repeat that explicit selection when the system is switched to the other mode.  It's straightforward to add the re-check, but it would be good to research whether there's another way that's more automatic.  We can't be the only people who want to dynamically switch between light and dark icons.
* Everywhere else, getting the various colors right depends on palette propagation, and palette propagation is blocked anywhere there's a style sheet set, whether or not the style sheet sets colors.  Commenting out "set style sheet" calls makes dynamic dark/light switching work for the affected subtree.

There's another small thing that I can't explain yet: a checked checkbox is black-on-blue, in either mode, but in dark mode when the window doesn't have focus the checkbox goes totally black.  (For a long time I thought that this was a dynamic-switch thing, but I was clicking on the Control Panel window to change modes, and *that* was what was causing the checkboxes to go black.)