openscad/openscad #4902: Have resolution option for "Export as Image"

j00ch, 2023-12-23:
Sometimes I'm designing a part which accidentally looks really pretty in preview. Me and some friends dubbed it OpenSCAD art.

[image]

I'd like to share these previews as such and i currently use the "Export as Image" option. The resolution however is too low to use in print or even digital platforms.

Sure, you could render an STL and use a different program to get a higher resolution image. Renders however do not have any color information. The preview is best.

I'd like to have an option where i can set the export image resolution.

nophead, 2023-12-23:
You can set the resolution from the command line. I set it to twice what I want and then scale it down by a factor of 2 using ImageMagic to get antialiasing.

tpimh, 2023-12-28:
An ideal solution for artistic renders would be to export a 3D model with color information, then import it in Blender to render. As far as I know, nobody is currently working on color export (#4671).

nophead, 2023-12-28:
You can add colours and textures afterwards in Blender. This was modelled in OpenSCAD and rendered in Blender.

[image]