openscad/openscad #4903: libhogweed.so.6 failed to reference __gmpn_cnd_swap

Hoek67, 2023-12-23:
Trying to compile OpenSCAD on Ubuntu (latest).

Get near the last step.

following instructions on https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Building_on_Linux/UNIX

This step has an error.
[image]

gets error...

[image]

I'd like to have a look at the export code and hopefully add a zip option with .obj and .mat in it.

I'm not a Linux person so tried a few things and it didn't fix it. (ie. install nettle)

Any help appreciated.

Seems GMP got broken somewhere.

kintel, 2023-12-23:
If you're not trying to do something very specific, the install guide recommends `./scripts/uni-get-dependencies.sh`, not `./scripts/uni-build-dependencies.sh`.

t-paul, 2023-12-23:
Also duplicate of #4741.

closed 2023-12-23 by t-paul