openscad/openscad #4909: Upgrade to Qt6

kintel, 2023-12-31:
This is a tracking issue for all work needed or wanted to be able to move OpenSCAD to Qt6.

TODO:
* [ ] Create a build-time Qt selection CMake flag
* [ ] First working Qt6 build on Linux
* [ ] First working Qt6 build on macOS
* [ ] First working Qt6 build on Windows
* [ ] Linux AppImage snapshot build
* [ ] Windows snapshot build
* [ ] macOS snapshot build
* [ ] Other Linux builds, including distro packages
* [ ] Go through and manually test each Qt-based feature on each target platform. This can be a bit of work, so we might want to structure this as checklist or smth.
* [ ] Update website to offer direct download of Qt6 builds
* [ ] Game pad support
* [ ] Remove (or hide) Qt5 downloads, and make Qt6 downloads the default

kintel, 2023-12-31:
@cjmayo FYI

kintel, 2023-12-31:
Changes to `CMakeList.txt` and `FindQt6QScintilla.cmake` to make OpenSCAD build on Qt6: #4908

Scopeuk, 2024-05-11:
As a note on the qtgame pad it looks like the maintainer decided it should be re-implimented at a lower level during the 5->6 jump but other things took precedence (https://www.reddit.com/r/QtFramework/comments/13uperb/comment/jm5y9yc/ ), another developer has done a straight port at https://github.com/pumphaus/qtgamepadlegacy which could be a work arround short term

AaronVerDow, 2025-04-23:
Qt6 package for Nix: https://gist.github.com/AaronVerDow/b945a96dbcf35edfc13f543662966534

kintel, 2026-01-18:
> > Linux distro packages as applicable
> 
> Directly related are only the packages we build on OBS: https://build.opensuse.org/package/show/home:t-paul/OpenSCAD Those will need to be updated. I'm not sure if we want to rate those as blocker though.

It would be nice to get those on Qt6 as well, where possible, just to cover the OpenSCAD users we service.

coryrc, 2026-01-18:
So we just need to get changes merged to master and OBS will run ~soon after?

coryrc, 2026-01-18:
We already "build and test" [`windows-latest qt6`](https://github.com/openscad/openscad/actions/runs/21106917199/job/60699753524) so what is left for "Windows snapshot build (MXE)"?

Is it like #5455 and some in-repo script needs to *default* to qt5?

coryrc, 2026-01-19:
Another question: if you were starting from scratch, would you add CircleCI or base everything on Github Actions?

Because it seems to me there's two duplicate workflows for building binaries and that's a waste of maintainer time.

coryrc, 2026-01-19:
> > I think you have that confused with cygwin. https://www.msys2.org/docs/what-is-msys2/ says:
> > > MSYS2 focuses on building native software built against the Windows APIs.
> 
> If you can help making the MSYS2 build become distributable, that would be cool. I have too little understanding of the nuances involved in this. Even if we continue using GitHub+CircleCI, having a unified build setup would save us some headaches.
> 
> The main challenge I can think of is to figure out how to pin packages so we're not subject to a rolling release when building patch-level releases. Might be possible through some sort of docker caching.

My first thought is to save the entire environment after building a release as an artifact, then if building a point release, load that environment first, skip pulling packages, and do an incremental build. If you do add packages you're susceptible to cascading dependencies no matter what approach you take.

But also, you have a single "stable" release every five years. Point releases aren't really a thing which need optimizing.

> There is, of course, some value in maintaining a Linux-centric toolchain (mostly the ability to build things locally without acquiring a Windows computer), but that value may be diminishing now that CI vendors offer Windows runners.

It bugs me to have redundant builds. Every push to a PR causes builds on CircleCI and Github, and nearly every build on CircleCI will never be used or accessed, while the GHA ones get tests run.

coryrc, 2026-01-19:
> We hope to dramatically increase the frequency: Patchlevel releases following a full release (bugfixes, UI tweaks, performance fixes) should be possible to roll out monthly or so, and we shou;d set ourselves up to be able to release full versions a fair bit more often than even 5 years.

It was literally my job to do this. I take the lessons, backed by evidence, from [DORA](dora.dev) to heart. And I've found the best way to do this is to start with the smallest thing you can release and make it as fast as possible. But that also was for server-side applications where I could actually deploy on my own schedule and get users on it immediately, so being able to deploy in an hour is meaningful. In this case even daily releases won't get significant traffic and we already do that. So I'm not sure the stack is currently the limitation.

DORA teaches us that trunk-based development results in a better product: fewer bugs, lower severity of bugs. It's why Firefox is on "version 143". Maintaining separate feature branches to release only some of the changes to the binary is an ultimate net negative to your efficiency, and so you'll do better by not releasing only some changes "(bugfixes, UI tweaks, performance fixes)" -- you should just release more often period. This *doesn't* mean "don't *ever* make a patch"; if you have an obvious targeted fix for something with no or painful workaround and you have lots of other changes, sure, backport it. But the effort you make to backporting is better spent making the next release even better and adding testing so you don't make such a screwup. And DORA's research confirms this across hundreds (maybe thousands by now?) of different kinds of organizations and cultures.

And that gets to my motivations for this request:

Let's release the MSYS2 build we already make on GHA as the official release.

It's a small change. It's something which can be iterated on quickly: all the dependencies are in this single repo and system, instead of spread across many systems and repos. It lets us eliminate a whole bunch of baggage: a half-dozen docker images and their build files, a bunch of circleCI config. There isn't any wasted effort because we haven't yet figured out qt6 for MXE yet.

kintel, 2026-01-19:
> > Stability of the file format is key, and this is the primary driver of OpenSCAD releases.
> 
> Which is better addressed by directly addressing that via regression testing. In the last five years, when has there been significant unintentional regression which wasn't quickly caught? Along the way there could have been more releases.
> 
It's not so much about regression testing, it's about planning how and when to introduce new language features or built-in modules, or other behavioral changes.

coryrc, 2026-01-20:
> It's not so much about regression testing, it's about planning how and when to introduce new language features or built-in modules, or other behavioral changes.

Which is currently managed by the feature system, which is the best way as all the code is being built and can't fall to being unbuildable (like #4478 is now).

coryrc, 2026-01-20:
> > There isn't any wasted effort because we haven't yet figured out qt6 for MXE yet.
> 
> Meaning what exactly?
> 
> Like changing `qtbase` to `qt6-qtbase` in?
> 
> https://github.com/openscad/docker-openscad/blob/d86594133c642887e2ad3aad5b06cf02ccf56f70/mxe/packages.lst#L9

https://github.com/openscad/docker-openscad/pull/29 proves this dismissive attitude was wrong. It's not simple, it's not documented, and basically only one person can do it without reverse-engineering multiple repos of redundant stuff.