openscad/openscad #4912: Code completion for modules in included files

hatalmasalma, 2023-12-31:
For me it seems like code autocompletion doesn't use module or function names from include() or use() files, only from the currently edited file.

I hope it would be easy to implement.

kberkut, 2024-01-09:
How did you enable auto-completion of the code?

hatalmasalma, 2024-08-17:
Seems like autocompletion is extremely basic, and only completes names from the currently edited file by pressing Ctrl+Space (if I remember correctly)

Since then I moved on from OpenScad because the rendering times became unbearable even in my simple projects.

closed 2024-08-17 by hatalmasalma

pca006132, 2024-08-18:
@hatalmasalma you can give the development snapshot a try, just enable the `manifold` experimental feature and the rendering time should improve by a lot.