openscad/openscad #4920: Editor reverts to light theme on every start

korpx, 2024-01-04:
Editor reverts to light theme on every start

Steps to reproduce the behavior:
1. Start OpenSCAD
2. Click on 'Edit'-menu, 'Preferences', 'Editor'
3. Change 'Color syntax highlighting:' to 'For Dark Background'
4. Exit OpenSCAD
5. Start OpenSCAD
6. 'Color syntax highlighting:' has reverted to 'For Light Background'

Expected behavior;
'Color syntax highlighting:' stays permamently set.

Screenshots;
[image]

OpenSCAD Version: 2024.01.03 (git b1160fa43)
System information: Microsoft Windows 10 (10.0.19045) x86_64 4 CPUs 15.88 GB RAM
User Agent: OpenSCAD/2024.01.03 (git b1160fa43) (Microsoft Windows 10 (10.0.19045) x86_64)

korpx, 2024-01-04:
Even if I import this before start the setting is reverted;

Windows Registry Editor Version 5.00

[HKEY_CURRENT_USER\SOFTWARE\OpenSCAD\OpenSCAD\editor]
"syntaxhighlight"="For Dark Background"

SongZihui-sudo, 2024-01-04:
You can try editing in vscode, it works fine

korpx, 2024-01-04:
The internal editor work fine too but this is a cosmetic bug introduced recently.

kberkut, 2024-01-06:
On linux, the theme setting is also reset. The version is similar.

Nikorasu, 2024-01-06:
Yup, this happens to every dev build I've tried after OpenSCAD-2023.12.28.ai17902-x86_64.AppImage, that was the last one that still kept it's Editor background setting.

majekw, 2024-02-15:
Same here.
Os: Linux Slackware 15.0.
It works ok on OpenSCAD-2023.11.14.ai16996-x86_64.AppImage
Color settings is lost on startup on OpenSCAD-2024.02.11.ai18430-x86_64.AppImage

korpx, 2024-02-25:
Still happens with 2024.02.22 (Windows 10 22H2)

t-paul, 2024-03-03:
Fixed via #5026.

closed 2024-03-03 by t-paul