openscad/openscad #4921: Emits "PulseAudioService: pa_context_connect() failed" to stderr every 30 sec

CrashCash, 2024-01-05:
**Describe the bug**
The nightly now emits "PulseAudioService: pa_context_connect() failed" when you start it, and every 30 seconds thereafter. I am not running pulseaudio.

**To Reproduce**
Simply start "openscad-nightly" in an xterm.

**Expected behavior**
No errors emitted.

**Code reproducing the issue**
N/A - no code needed.

**Screenshots**
N/A - easy to reproduce.

**Environment and Version info (please complete the following information):**
 - OS: Devuan Ceres (aka Testing)
 - System: Intel PC 64-bit
 - OpenSCAD Version: OpenSCAD version 2024.01.01.nightly installed from Debian package from the opensuse repository.

**Library & Graphics card information**
# openscad-nightly --info
QObject::startTimer: Timers can only be used with threads started with QThread
OpenSCAD Version: 2024.01.01.nightly (git 38e21ce)
System information: Linux 6.5.0-5-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.5.13-1 (2023-11-29) x86_64 Devuan GNU/Linux 6 (excalibur/ceres) 8 CPUs 31.28 GB RAM
User Agent: OpenSCAD/2024.01.01.nightly (git 38e21ce) (Linux x86_64; Devuan GNU/Linux 6 (excalibur/ceres))
Compiler: GCC "13.2.0" 64bit
MinGW build: No
Debug build: No
Boost version: 1_74
Eigen version: 3.4.0
CGAL version, kernels: 5.5.1, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.6.0
Qt version: 5.15.10
QScintilla version: 2.14.1
InputDrivers: 
GLib version: 2.78.3
lodepng version: 20210627
libzip version: 1.7.3
fontconfig version: 2.14.2
freetype version: 2.13.2
harfbuzz version: 8.0.1
cairo version: 1.18.0
lib3mf version: 1.8.1
Features: fast-csg, fast-csg-safer, fast-csg-debug, manifold, roof, input-driver-dbus, lazy-union, vertex-object-renderers, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output
Application Path: /usr/bin
Documents Path: /root/.local/share
User Documents Path: /root
Resource Path: /usr/share/openscad-nightly
User Library Path: /root/.local/share/OpenSCAD/libraries
User Config Path: /root/.config/OpenSCAD
Backup Path: /root/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /root/.local/share/OpenSCAD/libraries
  /usr/share/openscad-nightly/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  /root/.local/share/fonts
  /usr/local/share/fonts
  /usr/share/fonts
  /root/.fonts
  /usr/X11/lib/X11/fonts
  /System/Library/Fonts
  /Library/Fonts
  /root/Library/Fonts
  /usr/share/fonts/X11
  /usr/share/fonts/cMap
  /usr/share/fonts/cmap
  /usr/share/fonts/eot
  /usr/share/fonts/liberation-fonts-ttf-2.1.2
  /usr/share/fonts/opentype
  /usr/share/fonts/svg
  /usr/share/fonts/truetype
  /usr/share/fonts/type1
  /usr/share/fonts/woff
  /usr/share/fonts/xscreensaver
  /usr/share/fonts/X11/100dpi
  /usr/share/fonts/X11/75dpi
  /usr/share/fonts/X11/Type1
  /usr/share/fonts/X11/cyrillic
  /usr/share/fonts/X11/encodings
  /usr/share/fonts/X11/misc
  /usr/share/fonts/X11/util
  /usr/share/fonts/cmap/adobe-cns1
  /usr/share/fonts/cmap/adobe-gb1
  /usr/share/fonts/cmap/adobe-japan1
  /usr/share/fonts/cmap/adobe-japan2
  /usr/share/fonts/cmap/adobe-korea1
  /usr/share/fonts/eot/font-awesome
  /usr/share/fonts/eot/material-design-icons-iconfont
  /usr/share/fonts/opentype/font-awesome
  /usr/share/fonts/opentype/fonts-hosny-amiri
  /usr/share/fonts/opentype/ipafont-gothic
  /usr/share/fonts/opentype/ipafont-mincho
  /usr/share/fonts/opentype/linux-libertine
  /usr/share/fonts/opentype/roboto
  /usr/share/fonts/opentype/stix
  /usr/share/fonts/opentype/stix-word
  /usr/share/fonts/opentype/terminus
  /usr/share/fonts/opentype/unifont
  /usr/share/fonts/opentype/urw-base35
  /usr/share/fonts/svg/font-awesome
  /usr/share/fonts/truetype/adf
  /usr/share/fonts/truetype/arphic-bkai00mp
  /usr/share/fonts/truetype/arphic-bsmi00lp
  /usr/share/fonts/truetype/arphic-gbsn00lp
  /usr/share/fonts/truetype/arphic-gkai00mp
  /usr/share/fonts/truetype/baekmuk
  /usr/share/fonts/truetype/beteckna
  /usr/share/fonts/truetype/crosextra
  /usr/share/fonts/truetype/dejavu
  /usr/share/fonts/truetype/downloaded
  /usr/share/fonts/truetype/droid
  /usr/share/fonts/truetype/font-awesome
  /usr/share/fonts/truetype/freefont
  /usr/share/fonts/truetype/gentium
  /usr/share/fonts/truetype/gentium-basic
  /usr/share/fonts/truetype/glyphicons
  /usr/share/fonts/truetype/lato
  /usr/share/fonts/truetype/liberation
  /usr/share/fonts/truetype/liberation2
  /usr/share/fonts/truetype/libreoffice
  /usr/share/fonts/truetype/lyx
  /usr/share/fonts/truetype/material-design-icons-iconfont
  /usr/share/fonts/truetype/mathematica
  /usr/share/fonts/truetype/mathml
  /usr/share/fonts/truetype/msttcorefonts
  /usr/share/fonts/truetype/noto
  /usr/share/fonts/truetype/open-sans
  /usr/share/fonts/truetype/openoffice
  /usr/share/fonts/truetype/roboto
  /usr/share/fonts/truetype/ttf-bitstream-vera
  /usr/share/fonts/truetype/tuffy
  /usr/share/fonts/truetype/unfonts-core
  /usr/share/fonts/truetype/unfonts-extra
  /usr/share/fonts/truetype/unifont
  /usr/share/fonts/truetype/vlgothic
  /usr/share/fonts/type1/gsfonts
  /usr/share/fonts/type1/mathml
  /usr/share/fonts/type1/urw-base35
  /usr/share/fonts/woff/font-awesome
  /usr/share/fonts/woff/material-design-icons-iconfont
  /usr/share/fonts/X11/encodings/large
  /usr/share/fonts/opentype/roboto/slab
  /usr/share/fonts/truetype/roboto/slab

GL context creator: EGL (new)
EGL version: 1.5
PNG generator: lodepng

GLAD version: 2.0.4
OpenGL Version: 4.6.0 NVIDIA 525.147.05
GL Renderer: NVIDIA GeForce GTX 1060 6GB/PCIe/SSE2
GL Vendor: NVIDIA Corporation
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

**Additional context**
I'm not running pulseaudio so it'll never connect to anything

MWStonton, 2024-01-05:
Can't help with the code but was wondering if it occurs independent of the setting in 
Preferences - advanced - "Play sound notification on render complete"

CrashCash, 2024-01-05:
Yes, I already had "Play sound notification on render complete" unchecked. Good call, though.

I wonder if there's any options in Qt to change the sound engine or turn it off. It gives me something to Google.

CrashCash, 2024-01-06:
Yes, I found that lib and moved it to /tmp, and it didn't help.

/lib # grep -Rls "pa_context_connect() failed"
x86_64-linux-gnu/qt5/plugins/audio/libqtmedia_pulse.so
x86_64-linux-gnu/libQt5Multimedia.so.5
x86_64-linux-gnu/libQt6Multimedia.so.6.4.2
x86_64-linux-gnu/pulseaudio/libpulsedsp.so
x86_64-linux-gnu/qemu/audio-pa.so
x86_64-linux-gnu/libQt5Multimedia.so.5.15.10
x86_64-linux-gnu/libQt5Multimedia.so.5.15
x86_64-linux-gnu/libQt6Multimedia.so.6
(and pretty much the same for /lib64)

If I move the libQt5Multimedia/libQt6Multimedia libs, then the apps don't start because they can't load libraries. I didn't have any luck finding a Qt user setting.

CrashCash, 2024-01-06:
Hm... well I've used OpenSCAD for a while... (couple years) and this is the first time I thought I'd seen this. I usually render a couple dozen times on the way to a final model. Not arguing though.

Anyway, yeah, it's not a big deal, especially compared to the enormous volumes of barf that GIMP, Firefox, PrusaSlicer, etc output from GTK. I didn't know if it was an indication of a problem behind the scenes. 

Thanks for taking a look.

Also THANK YOU for the new measurement tools! After getting half a dozen transforms deep, it's nice to be able to say "what is that distance, again?"

triffid, 2024-01-10:
> Gentoo
> an experimental OpenSCAD build is a challenge though!

https://bugs.gentoo.org/908591 may interest you - seems like cmake install isn't actually installing some of the new libraries that openscad builds for some reason, but Gentoo's existing -9999 ebuild (which grabs git head) can be patched to "manually" install them.

CrashCash, 2024-01-11:
Hm. So I just updated to 2024.01.06.nightly, which also pulled in some libboost packages.... and....

No more pulseaudio emissions. Alrighty then. I guess this is no longer a bug!

closed 2024-01-12 by pca006132